WEC-Sim/WEC-Sim #279: From Nemoh to WEC-Sim

EstebanChOr, 2018-12-10:
Hello, 

I'm a tenderfoot at WEC-Sim. Although I have been reading the correspondent documentation,  I have some doubts regarding the **h5 file** and the expected **geometry files** in the case directory, so for the sake of time-saving I would appreciate some error-anticipating advices. 

To mention:
OS: Ubuntu 18.04
WEC-Sim: V3
Matlab: R2018b

At the time, I am putting together a case directory to simulate  a very simple free floating sphere using BEM results from **Nemoh**.  I have used the Matlab routine **axiMesh** for generating the mesh file. With aid of **Bemio function**  I expected to get the proper hydrodynamic data collected in a h5 file, but for reasons I can't clearly see, this file is never generated, despite the fact that no error is thrown and all hydro information is stored in a struct in Matlab workspace. Off course, I then used the 3 provided functions (contained in the writeH5 folder) to manually write the h5 file by inspecting the hydro struct, but it is kind of impractical. Furthermore, my question is (because I suspect it doesn't) if this file works exactly the same as the one generated (when the code works properly) by **Write_H5** or the last is more complete in some way.  

Now, regarding the geometry files, I understand  WEC-Sim code's demand on **.stl**  file, and at this point my question rises ¿What is the best-proven way to proceed considering I do not own a Rhino license? I have got some good, compatible hydrodynamic results derived from .dat files generated directly by axiMesh.m ¿Should I somehow create a .stl file that is consistent with current Nemoh's .dat mesh? or rather create a .stl file which I can later convert to Nemoh (.dat) format with **meshConvert.m** and repeat the previous  process. 

I would really appreciate any advice and updates

greetings to the good wave community,

Esteban.

gui-paredes, 2018-12-19:
Dear Esteban,
Regarding the last part of the question, unless you want a Froude-Krylov or non-linear simulation, you can use any .stl file you want, even with a totally different geometry, because it will only be used for visualisation purposes (I don't use it much). If you do want to create a geometry file, than you can use FreeCAD which exports to .stl (you have to save as .ast  instead of the using the option to save directly as .stl, and afterwards change the extension manually to .stl; I won't go into the reasons for this unless absolutely required). 
You said you generated the mesh file, but have you computed the hydrodynamic coefficients in Nemoh? I am asking because I didn't see you mention that step.

gui-paredes, 2018-12-19:
I use ubuntu, Nemoh and wec-sim too and there are some quirks you need to fix before being able to use them in linux.

crobarcro, 2018-12-19:
If using FreeCAD you can create nice meshes using the mesh workbench and exporting this directly as STL (much better if using the STL in calculations, not just for visualisation).

EstebanChOr, 2018-12-19:
@crobarcro, thanks!  is it possible to open in FreeCAD the mesh file created with axiMesh.m routine from Nemoh-MATLAB wrapper, and then export? 

@gui-paredes, first of all, thank you. That's quite a useful observation.

I did compute the hydrodynamic coefficients using Nemoh, so they are stored in the **Results** folder. Then using **bemio.m** I expected to get the hydrodata **.h5** file, getting all the required files to finally use WEC-Sim. 

Now, should I take into consideration some additional step in order to run WEC-Sim without the geometry file and thus, the visualization part? 

and also, What are this quirks you are talking about? I have not run WEC-Sim but I did have some problems running **Nemoh**, particularly with the graphic drivers and UNIX commands, these I have fixed.

adetosoafonja, 2018-12-30:
I apologize for using this issue to ask my own question it is because I have not gotten any reply for the one I raised.

I am currently new to Nemoh. I have created my mesh using a CAD program and saved it in Nemoh dat format. However, I would like to know how I can generate the hydrodynamic parameters I need to run Nemoh matlab routine.
Prior to now, I had used the mesh.m file to create simple geometry and this outputs the stiffness matrix and other files which I believe are input necessary to run the nemoh.m file.
The question now is I do use the mesh I had created in an external CAD program to carry out all the required calculations.

kmruehl, 2019-01-08:
@EstebanChOr Thank you for you question. 

1. Can you provide us your NEMOH run and the BEMIO file for your floating sphere? When running BEMIO for NEMOH, we need the following files [listed on the website here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio): 

**Read_NEMOH: Reads data from a NEMOH working folder**
    `hydro = Read_NEMOH(hydro, filedir)`
- hydro – data structure 
- filedir – NEMOH working folder, must include:
-- Nemoh.cal
-- Mesh/Hydrostatics.dat (or Hydrostatiscs_0.dat, Hydrostatics_1.dat, etc. for multiple bodies)
-- Mesh/KH.dat (or KH_0.dat, KH_1.dat, etc. for multiple bodies)
-- Results/RadiationCoefficients.tec
-- Results/ExcitationForce.tec

kmruehl, 2019-01-29:
@EstebanChOr does this resolve your issue?

closed 2019-02-04 by kmruehl

abhisheks865, 2020-11-10:
Hello there,
I was trying to running the NEMOH code but unable to generate the hydrostatic.dat and kh.dat file after running it. Can somebody here help me to understand how can I generate these files so that I can use further use these files for Bemio.

gui-paredes, 2020-11-10:
Dear @abhisheks865 You will need to provide more details on your case for us to be able to help. You might also consider submitting your query at NEMOH's forum, as it is more likely for you to get the proper help there, if it is not related to any WEC-SM - Nemoh interfacing

abhisheks865, 2020-11-10:
Hello @gui-paredes thanks for responding. I am writing it here as my queries is related to NEMOH-WEC-Sim Interaction. In order to run Bemio, It needs hydrostatic.dat and kh.dat file as stated by @kmruehl above.
I tried to run the WEC-Sim BEMIO code (bemio.m) for Nemoh. It has four files in mesh folder (WEC-Sim-examples-BEMIO-NEMOH-RM3-Mesh) called hydrostatic_0.dat, hydrostatic_1.dat, KH_0.dat  and KH_1.dat so it is running fine. But when I am trying to run NEMOH for my object then it is not creating above four files and due to that I am unable to run the code on my geometry. Is there any change needed in Nemoh.cal or Mesh.cal file to generate these files or somewhere else. I am sorry but I am just a beginner and finding difficulty in running WEC-Sim with NEMOH.

okhsallam, 2021-02-21:
I have the same issue of not finding the Hydrostatic.dat and KH.dat 
Did you resolve this issue ?