WEC-Sim/WEC-Sim #283: modifying rigid block

adetosoafonja, 2019-01-10:
I would like to modify the rigid body block to accommodate dynamics of a body that is in my floater. However, I would need the to know what the the responses stand for. I know x,y,z is the surge, sway and heave. I need to know what is the pitch, sway and roll.

adetosoafonja, 2019-01-12:
I have been able to identify the motions. However, my project is to modify the rigid block to include the dynamics of a anti roll passive U-tank. Which means i need to make the rigid body(floater) affect the motion of the anti roll tank in pitch and the U-tank should also feed back and provide forcing terms on the floater. 
After implementation I got this error when carried out the simulation

Error using simscape.engine.sli.internal.solverErrorFcn
Simulink cannot solve the algebraic loop containing 'floater_ART/Global Reference Frame/Solver Configuration' at time 0.35500000000000004
using the LineSearch-based algorithm due to one of the following reasons: the model is ill-defined i.e., the system equations do not have
a solution; or the nonlinear equation solver failed to converge due to numerical issues.
 To rule out solver convergence as the cause of this error, follow either of the suggested actions. If the error persists in spite of the
 following changes, then the model is likely ill-defined and requires modification.

Error in simscape.engine.sli.internal.solverErrorFcn

Error in wecSim (line 257)
sim(simu.simMechanicsFile);
Suggested Actions:
    • Switch to TrustRegion-based algorithm using
 set_param('floater_ART','AlgebraicLoopSolver','TrustRegion') - Fix
    • Reduce the fixed step size. The fixed step size parameter can be accessed from 'Configuration Parameters > Solver > Additional
    options > Fixed-step size' - Open

yuyihsiang, 2019-01-15:
Roll, Pitch and Yaw are 4, 5, and  6

To solve algebraic loop, you may want to consider putting a delay block in the loop.

Yi-Hsiang

closed 2019-02-04 by kmruehl