WEC-Sim/WEC-Sim #286: reading NEMOH outputs for CAD-generated geometries

kwadl, 2019-02-19:
It seems that the NEMOH workflow only generates hydrostatics outputs (Hydrostatics.dat and KH.dat) when NEMOH's Mesh executable is run (on a coarse 2D representation of the geometry?) to generate NEMOH's input mesh. It looks like these files are not generated when you take the CAD->Meshmagick->NEMOH preProcessor->NEMOH Solver->NEMOH postProcessor route, which I am using successfully to generate all the other outputs from NEMOH. Because BEMIO requires the hydrostatics files to generate its HDF5 input file, this seems to put a real damper on the CAD->WEC-SIM pipeline. Does anyone know how I can get around this?

kwadl, 2019-02-23:
Okay, so I think I have a way through by using meshmagick's hydrostatics functions. The attached toolset for a Windows environment seems to let me go from a CAD generated STL file with z-plane=water plane at equilibrium, through to h5 + WEC-SIM STL file. To test, set up an environment variable PROTEUS_TOOLS pointing to the top level directory of the extracted contents, and run .\prep-geometry\test_geometry\process.bat. One thing I'm not 100% sure of despite reading #129 is whether there are any issues with using a geometry having z-plane=water plane as input to NEMOH, but then using an STL with origin at CoG for WEC-SIM. Is this okay, or does it invalidate the simulation somehow?

[proteus_tools.zip](https://github.com/WEC-Sim/WEC-Sim/files/2897343/proteus_tools.zip)

yuyihsiang, 2019-03-04:
I believe what you are saying is that you will have a geometry (mesh) in nemoh that has an origin at the free surface and has the STL with origin at CoG for WEC-SIM. This is what we need to have using nemoh+WEC-Sim. If this is what you are doing, then there is no issue at all.

Yi-Hsiang

closed 2019-03-19 by kmruehl