WEC-Sim/WEC-Sim #287: Question on how to install WEC-Sim

rjan727, 2019-02-23:


yuyihsiang, 2019-03-04:
Here is an example of my startup.m file. Please note that mine is for OSX. If you have a window machine you need to switch "/" for "\".

wecSimPath = '/Users/yyu/Documents/Github/WEC-Sim/source';
addpath(genpath(wecSimPath));
clear wecSimPath;

kmruehl, 2019-03-05:
@rjan727 you will need to copy this code and place it in your `startup.m` file specifying your local `wecSimPath` as @yuyihsiang has shown. This adds the WEC-Sim source directory (and it's sub-directories) to your MATLAB path. Your `startup.m` file is located in your MATLAB Startup Folder. For more information about the MATLAB Startup Folder and the  `startup.m` file, refer to the following MATLAB guidance:

https://www.mathworks.com/help/matlab/matlab_env/matlab-startup-folder.html
https://www.mathworks.com/help/matlab/ref/startup.html

rjan727, 2019-03-05:
Thank you for your help! I contacted technical support from Mathworks and successfully added WEC-Sim to my startup.m file path.

sugarl14, 2020-01-14:
@rjan727 Hey there! I seem to be having a similar issue. Would you mind emailing me and we can possibly set up a time to talk about it together? I'm sure I'm going through similar issues that you may have already answered! My info is SugarL14@students.ecu.edu

Thanks!!

Yifan2020, 2020-01-23:
Excuse me, can you tell me how you add WEC-Sim to your startup.m file path? Thanks.

Yifan2020, 2020-01-23:
@rjan727 My email address is yifanluo2018@gmail.com. can you send how you add WEC-Sim to your startup to your startup.com file path? thanks.

nathanmtom, 2020-01-23:
@Yifan2020 can you confirm you have followed the steps listed on the "Getting Started  http://wec-sim.github.io/WEC-Sim/getting_started.html ? Perhaps you can clarify if you are unable to open the startup file or do not understand which lines of code to include?

Yifan2020, 2020-01-23:
@nathanmtom 
I am on the step:
"Copy the code in the 'wecSimStartup.m' shown above and paste it into the 'startup.m' file located in the MATLAB Startup Folder. Set the 'wecSimPath' variable to the '$wecSim' folder and open the 'startup.m' file by typing 'open startup.m' into the MATLAB Command Window:"
I do not know where to find the startup.m or create the startup.m. I tried to create the 'startup.m' in the 'source' folder but it did not show in the Simulink Library Browser.

crobarcro, 2020-01-23:
at the matlab command prompt type 

`userpath`

whatever location this tells you then, put startup.m there

This is explained here: https://uk.mathworks.com/help/matlab/ref/startup.html

Yifan2020, 2020-01-24:
@crobarcro Thank you very much. I made it.

closed 2020-01-31 by nathanmtom