WEC-Sim/WEC-Sim #289: WEC-Sim Read Input File

AmirH7076, 2019-03-19:
Hello   
When I run WEC-Sim, I get the following error in MATLAB: 
Error using  < 
Not enough input arguments.

Error in wecSim (line 88)
        if h5Info.bytes < 1000
what is the problem? Please help me.
Thanks alot.

kmruehl, 2019-03-20:
@AmirH7076 this is an error that checks the size of the `h5` hydro data file, and flags and error if the `h5` file is too small (an indicator that it's not the correct file).  This means that you do not have the correct `h5` file in your case directory, and will either need to download it manually, or install git-lfs. Refer to the note for more info: http://wec-sim.github.io/WEC-Sim/getting_started.html#download-wec-sim

AmirH7076, 2019-03-21:
thank you.
How can I create .h5 File ? Its unclear for me. in BEMIO there is a line that say -Write_H5(hydro)- but where location does this save?

kmruehl, 2019-03-21:
@AmirH7076 information on how to generate the `h5` file from BEMIO is available on our website here: http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio. We also have a set of BEMIO tutorials provided with the WEC-Sim source code in the [WEC-Sim/tutorials/BEMIO](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO) directory. When you run BEMIO the `h5` is saved in the same directory as the BEMIO run file. When you run the BEMIO tutorials you will see an h5 generated in the tutorial directory.

AmirH7076, 2019-03-21:
Thank you so much.

sricharanvvs, 2019-06-07:
Following up the error, I am able to generate IRFs and graphs to visualize and thus the .h5 file. But why does it now become a garbage file showing this error.

kmruehl, 2019-06-07:
@sricharanvvs What error are you seeing?

sricharanvvs, 2019-06-07:
@kmruehl 
when i try to execute wecSim after geneating .h5 and editing wecSimInputFile.m  , I get 2 types of errors:
1: When I use .h5 file (which was generated using 2015b) in 2017b version (windows, with old WECSIM 2017) It says file is very large.
2: When I generate .h5 file in 2017b(windows), Ill get h<1000  binaryetc 
But I have read your answers in couple of threads. I generated this many times using bemio.m but no use. You also said the file size is too small etc or corrupted. Then why I am able to see all the hydrodynamic data graphs at the end of h5 file generation?
**I have also read about GITLFS but I really not sure how to use: The information in website is bit tricky to a naive. Is there any video reference for storing the files and retrieving back?** I have attached the Git cmd error

[Github.txt](https://github.com/WEC-Sim/WEC-Sim/files/3267640/Github.txt)

cadu55br, 2019-06-25:
Dear all,
I created a Matlab script (VCF_190625_1800_R5H6_axiMesh_Nemoh_Bemio.m) in order to generate the WEC-Sim input files. It seems this step was successfully.
I renamed the “axisym.dat” file to “float.dat”, and stored it in the “geometry” directory. (Note: Since I am creating a linear model, I am not using a *.stl file, using *.dat Nemoh file instead. Is it OK?).
I stored the “VCF_190625_1800_R5H6.h5”, file created by BEMIO, in the “hydroData” directory.
I adapted the original “RM3.slx” Simulink Model, considering one body only (a Point-Absorber WEC), and renamed it as “VCF_190625_1800_R5H6.slx”.
I stored everything in the case directory (VCF_190625_1800_R5H6).
Finally, I run the wecSimInputFile.m, and then I run “wecSim” from the command window, but I got the following error message:

WEC-Sim Read Input File ...   
Index exceeds the number of array elements (1).
Error in wecSim (line 86)
        h5Info = dir(body(ii).h5File);

Could you please inform what is the problem?
Thanks in advance!
Best regards,
Carlos

PS: I have attached PrtScr of my MATLAB screen.

[image]

cadu55br, 2019-07-04:
Hello again!
Regarding to my comment above, I am attaching the inputs & outputs (excluding 'Normalvelocities.dat', due to the file size) of the Matlab script ('VCF_190625_1800_R5H6_axiMesh_Nemoh_Bemio.m'), the Simulink model 'VCF_190625_1800_R5H6.slx' and the script 'wecSimInputFile.m'.

I hope this information can help to solve the error.
Thanks in advance!
Best regards,
Carlos

[inputs&outputs (excl. Normalvelocities.dat).zip](https://github.com/WEC-Sim/WEC-Sim/files/3360556/inputs.outputs.excl.Normalvelocities.dat.zip)

[wecSimInputFile.zip](https://github.com/WEC-Sim/WEC-Sim/files/3360560/wecSimInputFile.zip)

cadu55br, 2019-07-10:
In addition to my comments above, it seems the problem is within Hydrodynamic Data File generated by BEMIO: 'hydroData/VCF_190625_1800_R5H6.h5'

I have attached two text files with the outputs from MATLAB functions "h5disp" and "h5info", with information about the hydroData file.

Best regards,
Carlos

[h5disp('VCF_190625_1800_R5H6.h5').txt](https://github.com/WEC-Sim/WEC-Sim/files/3378923/h5disp.VCF_190625_1800_R5H6.h5.txt)

[h5Info = dir(body(ii).h5File).txt](https://github.com/WEC-Sim/WEC-Sim/files/3378924/h5Info.dir.body.ii.h5File.txt)

kmruehl, 2019-07-15:
@sricharanvvs in regard to your question about BEMIO and generating `*.h5` files. 
1. Please pull the latest version of WEC-Sim from GitHub: https://github.com/WEC-Sim/WEC-Sim
2. When running BEMIO it is possible to see the plots using the `Plot_BEMIO` function and not have the `*.h5` file written by the `Write_H5` function. The error you got means that the `*.h5` file was not written properly. Here is a webinar on running BEMIO: http://wec-sim.github.io/WEC-Sim/webinars.html#webinar-1-bemio-and-mcr

closed 2019-07-30 by kmruehl

chakramtalla, 2024-03-11:
i am getting the same error .h5 error, Ill get h<1000 binaryetc ( Matlab2023b)