WEC-Sim/WEC-Sim #291: NEED your urgent Guidance WEC-sim

muzammilhuss, 2019-04-06:
Hello, Dear members,
I successfully added up WEC-sim blocks in a SImulink Library. I am new to matlab. but not very expert yet.
My issue is that, since 3 days of struggle, I found that, this code does not run smoothly in 2018b. I tried it on 2018a as well. its gives same errors.
the error is shown in screen shot. Please guide me, how to resolve it. I am just worried. Thanks
[image]

yuyihsiang, 2019-04-08:
I believe you did not have the h5 files in the correct folder.
The error message is saying that the file it was looking for is basically 0k. 
You have to download and REPLACE the files in the hydroData folder inside each case (e.g., inside the RM3 and OSWEC tutorials).

YI-Hsiang

muzammilhuss, 2019-04-09:
Thank you so much Yl-Hsiang. Problem solved. simulation is able to run smoothly. 
But, I would like to know that, although the simulation runs, but I am not able to use PTOSim_Lib, generator blocks from Matlab library tree which have been added by me. I mean, when I give some constant value and add it into new project, It asks me for ptosim.pmRotaryGenerator(1) . Which files I have to run? so that i can use those blocks for my study.
Also, for classes, mooring, PTO etc ....e.g., 
if,
constraint(#) = constraintClass('<Constraint name>')
then,
what could be the constraint name? Do I must have to give it a name? if yes, then what that is supposed to be? Give me an example.
I already studied from the website, and I did not get the idea of it. How to do?
[image]
My major purpose if use build in WEC ( library) block presented by you.... 
Thank you

kmruehl, 2019-04-11:
@muzammilhuss this error is likely due to the fact that there are several PTO parameters that are defined in the `ptoSimInputFile.m`. The `ptoSimInputFile.m` must be fined to define there parameters. For examples of these files, refer to the WEC-SIm_Applications repository, in the PTO-Sim directory, [here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim).  And for more information about PTO-Sim, refer to the documentation [here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#power-take-off-pto-sim).

kmruehl, 2019-04-18:
@muzammilhuss did this resolve your issue?

muzammilhuss, 2019-04-19:
Hi thanks. This is the error. Well, its so complex. every time new error comes. how to fix this library now? 
[image]

kmruehl, 2019-04-22:
@muzammilhuss it looks like there were a few broken library links in the PTO-Sim example files on the WEC-Sim_Applications repository (https://github.com/WEC-Sim/WEC-Sim_Applications). I just updated the library links for the`Rotary Generator Lookup Table` for the case you were running, and now the `OSWEC_Hydraulic_Crank_PTO` case should run without error. Please pull the latest version of the WEC-Sim_Applications repository to resolve this issue. 

[image]

[image]

muzammilhuss, 2019-04-25:
I think this is a bug... which needs to be fixed. How to do?
[image]

kmruehl, 2019-04-25:
@muzammilhuss It looks like you're running WEC-Sim from the Source Directory, not the Case Directory. To run WEC-Sim, you must be in the Case Directory, and execute WEC-Sim by typing `wecSim` into the command window. 

The error related to the convolution integral time is because `simu.CITime` is larger than the impulse response function for the available hydrodynamic data. This can be resolved by reducing `simu.CITime`  in the `wecSimInputFile.m`

For more information on running WEC-Sim, refer to documentation here:
http://wec-sim.github.io/WEC-Sim/overview.html#running-wec-sim

kmruehl, 2019-05-06:
@muzammilhuss does this resolve your issue?

muzammilhuss, 2019-05-07:
Thank ypou very much. some fo the issues have been resolved successfully. But I think this error is about something else. this error comes when I runs "wecSiminutfile.m" of irregularCIC and RM3_MoorDyn...

[image]
[image]

kmruehl, 2019-05-07:
@muzammilhuss I was able to run the RM3_MoorDyn simulations with `waveClass('etaImport');`, `waveClass('regularCIC')` and `waveClass('irregular')` on my machine without issue. There is not a waveClass called irregularCIC, see list of wave classes here: http://wec-sim.github.io/WEC-Sim/code_structure.html#wave-class. 

[image]

I believe the issue is that you are running WEC-Sim simulations from the **$SOURCE** directory, not the **$CASE** directory. Only the **$SOURCE** directory should be in your MATLAB path, and you must be in the **$CASE** directory to run the RM3_MoorDyn simulation. 

For me, the **$CASE** directory is: _C:\Users\kmruehl\Documents\GitHub\WEC-Sim\WEC-Sim\WEC-Sim_Applications\RM3_Mooring\RM3_MoorDyn_, and my **$SOURCE** directory is: _C:\Users\kmruehl\Documents\GitHub\WEC-Sim\WEC-Sim\source_.

closed 2019-05-30 by kmruehl