WEC-Sim/WEC-Sim #293: Gap resonance in multi-WEC

Tudore, 2019-04-15:
Hi, just a reminder,

For multibody simulation in WAMIT, Nemoh, etc, there might be trapped mode when there exist narrow gaps. They can be non-physical and cause some problem in the WEC-Sim simulation.

To solve this problem, there can be several ways.
1. Introduce gap lid in WAMIT, this will suppress the gap resonance.
2. Introduce free surface damping term in HydroStar/WADAM, both results have not been used for WEC-SIM.
3. Try to skip the frequency around the gap resonance.

I am not sure whether Nemoh has implemented this function or not (to my understanding, no).
So better to be careful.

yuyihsiang, 2019-04-22:
Hi Tudore,

Method 1 could be an interesting solution. Is there a reason why you need to remove those gap-resonance hydrodynamic coefficients? I agree it will be great to remove them. However, the influence of those gap-resonances on the WEC response maybe negligible if "regularCIC" or "irregular" waves are used depending on cases and on how bad it is. I am just trying to see if you really need to go down that pass and see if there is a way to get around the issue and we can help.

Let us know what you have found out. Regarding Nemoh, I don't think they have this functionality. 

Best,

Yi-Hsiang

Tudore, 2019-04-28:
Regarding this problem, I will provide a simple case to show more details. 
The gap resonance is a problem but is overestimated in potential flow (added mass and damping can be very large if the frequency is close to the resonance frequency). If we have that jump in the hydrodynamic coefficient, it will cause the Retardation function decays very slow (lack of damping). The gap resonance also influence the added mass at infinite frequency. 
I will come back later. Too busy this month...
Thank you!

yuyihsiang, 2019-05-21:
Sure, we will be looking forward to the file.

Tudore, 2019-05-26:
Hi, Yihsiang, 

I made a simple case which is a barge with a moonpool inside. Although it is slightly different from the side by side case, the fundamental theory is similar. My barge has a dimension L*H*D = 100*50*4.

The added mass and damping are plotted as follows,

[image]
[image]

There exist quite several peaks on the added mass and potential damping. They are induced by resonance in the moonpool.

These sharp peaks are not physical. They are just there because no viscous damping is considered in the potential flow solver. In such a case, the IRF is calculated as follow,
[image]

Theoretically, there should not be any reduction on the IRF as there is no damping term. Due to the coarse numerical simulation frequency step, the peak is not infinite. So it behaves as a decaying trend. I believe you can notice after 100s, the IRF is still oscillating. 

These will not influence a simulation when,
1. if your simulation has very coarse frequency interval (e.g., 0.1rad/s, 20 frequencies...) and if it is lucky enough that those resonant frequencies can be avoided.
2. You are simulating bodies far away from each other or a single body.
3. introduce viscous damping to suppress the unrealistic resonance.

yuyihsiang, 2019-05-26:
Thank you, I will take a look this week.

Regarding mean drift force, I agree it is proportional to A^2. However, for irregular waves, A is not the amplitude. We actually use the square root of waves.A. Therefore, we have 

    C0 = bsxfun(@times,A*WaveSpread(ii),dw);
    C1 = sqrt(bsxfun(@times,A*WaveSpread(ii),dw));
    D0 =bsxfun(@times,squeeze(fExtMD(ii,:,:)),C0);
    D1 =bsxfun(@times,squeeze(fExtRE(ii,:,:)),C1);

C1 is equal to C0^2. Let me know if this makes sense or not.

Best,

Yi-Hsiang

yuyihsiang, 2019-05-28:
Thank you for sharing your case. It looks very interesting, and I agree that the coefficients consist of many sharp spikes. I think modeling barge with a moonpool inside in those BEM model can be tricky. 

Here are a few thoughts:
- I wondering if you have sufficient panel on the body. Sometimes we see those oscillating IRF if the panel resolution is insufficient, particularly for high-frequency cases.
- There should be some reduction on the IRF due to the radiation damping. 
- A paper by Newman last year may provide some information http://www.wamit.com/Publications/moonpool_rev.pdf

Best,

Yi-Hsiang

closed 2019-07-15 by kmruehl

Tudore, 2019-08-01:
> Thank you for sharing your case. It looks very interesting, and I agree that the coefficients consist of many sharp spikes. I think modeling barge with a moonpool inside in those BEM model can be [...]

Hi, Yi-Hsiang,

Just come back from the ISOPE conference and long summer vacation.

Regarding the drift force, it would be better if we consider Newman approximation in the calculation. Now only the diagonal term has been included in the calculation if I understand it correctly.

Regarding the gap resonance,
1. The first point about the panel number. Actually, I am using HOBEM and the RAOs has been verified in the FQ domain.
2. For the moonpool case and sloshing case, there may not be enough radiation(especially for sloshing). Since the water in the tank or moonpool is enclosed. There should be some viscous damping. 
3. Thanks for the reference. I will look into it these days.

Regards,
Zhang Chi