WEC-Sim/WEC-Sim #294: Inclusion of structural flexibility in Design load analysis

Tudore, 2019-04-15:
I have read a paper published by NREL. "Inclusion of structural flexibility in Design load analysis..."
In this paper, it mentioned WAMIT has been used to calculate the generalized mode response. And time domain solution is given by WEC-SIM. Before that, the structural modes are imported into WAMIT. May I know if you have any material on how to achieve this?

yuyihsiang, 2019-04-22:
Hi Tudore,

If you are asking how the structural modes are imported into WAMIT, you will have to check the WAMIT user manual Chapter 9 "GENERALIZED BODY MODES". For WEC-Sim implementation, the latest BEMIO has the capability to read in the WAMIT output with the generalized body mode and solve the system equation with the additional flexible DOF using the Flex body block in the WEC-Sim Lib. Note that the one in the GitHub is still a beta version and we can still checking it as we speak.

Thanks,

Yi-Hsiang

Tudore, 2019-04-28:
Hi, Yi-Hsiang,

Thanks for your reply. Yes, I do have tried the function in WAMIT to include generalized modes in the calculation. It is quite a interesting function. I will try to see how it works in time domain in WEC-SIM. 

Thanks!
Tudore

closed 2019-04-29 by kmruehl