WEC-Sim/WEC-Sim #295: BEMIO Reads AQWA Output Files

HondayMax, 2019-04-16:
Hello 
I am a beginner of WEC-Sim. I have some trouble loading AQWA output files.  
I want to calculate a simple  RM3 in Aqwa , and I get the AQWA output file, but when I use BEMIO, I make the following mistakes: 
The assignment cannot be performed because the size on the left is 6-by-1 and the size on the right is 1-by-5. 
error in  Read_AQWA (line 201)
                hydro(F).sc_ph((kdiff-1)*6+1:kdiff*6,j,i) = tmp2;  % phase of scattering force
error in  bemio (line 4)
hydro = Read_AQWA(hydro,'aqwa_example_data.AH1','aqwa_example_data.LIS');

[rm3.zip](https://github.com/WEC-Sim/WEC-Sim/files/3083904/rm3.zip)

kmruehl, 2019-04-22:
@HondayMax I was able to reproduce this error. It looks like there is an issue with the `READ_AQWA` script's ability to import the AQWA run you provided. The `READ_AQWA` script was developed to import AQWA data for a large number of wave frequencies, where the `*.LIS` file has one table per wave heading. The AQWA data you provided has 20 wave frequencies per wave heading, and the `*.LIS` file has multiple headings in each table. This makes the `*.LIS` file from AQWA pretty messy and complicated to import. 

For example, here's the `*.LIS` data from the [BEMIO AQWA tutorial](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO/AQWA/Example). Each wave heading has it's own `DIFFRACTION FORCES-VARIATION WITH WAVE PERIOD/FREQUENCY` table:
[image]

And here's the `*.LIS` data from the BEMIO AQWA run you uploaded. Each `DIFFRACTION FORCES-VARIATION WITH WAVE PERIOD/FREQUENCY` table has multiple wave headings:
[image]

**Based on this information, I recommend one of the following options:**
1. Rerun AQWA with a larger number of wave frequencies per wave direction. This should solve the `READ_AQWA` import issue by putting each wave heading in its own table. Moreover, WEC-Sim's accuracy in modeling WEC dynamics is greatly improved by a BEM solution with a larger number of wave frequencies (e.g. improve the calculation of the impulse response functions, response near resonance, etc.)
2. Modify the `READ_AQWA` script to manually define the correct line numbers to import data from the  `DIFFRACTION FORCES-VARIATION WITH WAVE PERIOD/FREQUENCY` table for each wave heading. I've provided some information on how to do this for you below. 

The issue occurs on line 190 where the `READ_AQWA` parser defines the next line of wave heading data to load. If each wave heading has its own table in the `*.LIS` file, the formula on line 190 is correct. However, the AQWA data you provided has multiple wave headings in each table, so you will need to manually define the line to import for each wave heading (`j=1:hydro(F).Nh`)
[image]

The `DIFFRACTION FORCES-VARIATION WITH WAVE PERIOD/FREQUENCY` table is correctly imported for the first wave heading of -180deg, corresponding to `raw2{35194}`. But the line imported for the second wave heading of -156deg is `raw2{35223}`, and it should be `raw2{35214}` (see below). To resolve this you'll need to modify line 190 in the `READ_AQWA` script to load the correct lines of the `*LIS` data. 
[image]

Hope this helps, sorry for being verbose.

HondayMax, 2019-04-23:
@kmruehl  I increased the wave frequency.  BEMIO could run and I got .h5 file.However, when I run it with  files under the path WEC-Sim_Applications\PTO-Sim\RM3_DD_PTO, the following errors occurred :

Misuse of h5readc 
The HDF5 library encountered an error and generated the following stack trace information: 

    H5G_traverse_real    component not found
    H5G_traverse         internal path traversal failed
    H5G_loc_find         can't find object
    H5Dopen2             not found

Error in h5read (line 58)
[data,var_class] = h5readc(Filename,Dataset,start,count,stride);

Error in bodyClass/readH5File (line 90)
            obj.cg = h5read(filename,[name '/properties/cg']);

Error in wecSim (line 92)
        body(ii).readH5File;

      I looked at the H5 file and found that there was only one body 1 and no body 2, but I built float and plate models in AQWA. 
[image]

[h5.zip](https://github.com/WEC-Sim/WEC-Sim/files/3107274/h5.zip)

kmruehl, 2019-04-23:
@HondayMax I'm glad that increasing the number of wave frequencies resolved the data import issue. The `*.LIS` file format from AQWA changes depending on the BEM input parameters and AQWA verision. Making a robust `READ_AQWA` script for importing different AQWQ data has proved challenging. 

I haven't been able to figure out why the script is not properly reading in the number of bodies from your AQWA run. I'll look into this a little more later. But a quick fix is to manually define the number of bodies in the `READ_AQWA` script. The number of bodies is defined on line 49 of the `READ_AQWA` script, simply change its value to 2 (see below). By doing this I was able to generate the attached `h5` file. 

[image]

[aqwa_example_data.zip](https://github.com/WEC-Sim/WEC-Sim/files/3108513/aqwa_example_data.zip)

HondayMax, 2019-04-24:
@kmruehl  Manual definition works well, and I can calculate models of different sizes just like RM3 in the tutorial. Thank you very much for your guidance. I believe WEC-Sim will get better and better. 
                                                                                                                                             
                                                                                                                                                 Thanks again,
                                                                                                                                                  Honday Max

closed 2019-07-15 by kmruehl

sricharanvvs, 2020-04-16:
Hi,
Unlike the above error, I am getting this something like this:

Index exceeds the number of array elements (0).

Error in Read_AQWA (line 178)
                    ind = tmp1(1:2); tmp1(1:2)=[];

Error in bemio (line 4)
hydro = Read_AQWA(hydro,'ANALYSIS.AH1','ANALYSIS.LIS');

I have run it for 100 frequencies. I have compared with the rm3 and my old files. I havn't found any change like the ISSUE #295. Both seems to be the same. Sad part is I have another .AH1 and .LIS file with me which doesnt have this problem in generating .h5.

I am attaching the following files:
1. New Error File: 'ANALYSIS.AH1','ANALYSIS.LIS'
2. Old correct file: 'Correct.AH1', 'Correct.LIS'

I tried some changes in line 178, the error then changes to line showing : Unable to perform assignment because the size of the left is 6-1 and right is 5-1 etc.
Both the attached files were run in old and the latest version of READAqwa.m files.
[ANALYSIS.zip](https://github.com/WEC-Sim/WEC-Sim/files/4484437/ANALYSIS.zip)
[Correct.zip](https://github.com/WEC-Sim/WEC-Sim/files/4484439/Correct.zip)