WEC-Sim/WEC-Sim #296: Getting the pitch, roll, and yaw

johnallen24, 2019-04-19:
In the user defined functions and response class for the RM3 example, position, velocity, and acceleration can be plotted for the two bodies. Is there any way to also get the pitch, roll, and yaw and plot those as well?

yuyihsiang, 2019-04-22:
In the output class, you should be able to find all the information you need for each body and PTO. Please also note that the RM3 example was done using 3DOF constraint, which does not allow any motion in sway, roll, and yaw.

Yi-Hsiang

johnallen24, 2019-04-22:
Is it possible to make the RM3 example have 6DOF?

yuyihsiang, 2019-04-26:
Hi johnallen24

You can replace the 3DOF joint with the 6DOF joint from the WEC-Sim lib, and it should work. Please note that the way the mooring was specified does not provide any restoring stiffness in yaw. 

Yi-Hsiang

closed 2019-05-06 by kmruehl