WEC-Sim/WEC-Sim #297: Ability to have waves in multiple directions

johnallen24, 2019-04-21:
The default of the simulation only has waves moving in one direction. Is it possible to have waves moving in multiple directions, more like the actual surface of an ocean?

yuyihsiang, 2019-04-22:
WEC-Sim does allow a specification of wave directional spreading
https://wec-sim.github.io/WEC-Sim/advanced_features.html#wave-features

Yi-Hsiang

johnallen24, 2019-04-22:
I try to change the waveDir property to have two directions in the wecSimInputFile for the RM3 example. 
`waves.waveDir = [0;45]`

However, I get this error

> Unable to perform assignment because the size of the left side is 1-by-1 and the size of the right side is 2-by-1.
> 
> Error in waveClass/waveElevReg (line 629)
>                     obj.waveAmpTime1(i,2)   = obj.A*cos(obj.w*t-obj.k*(obj.wavegauge1loc(1).*cos(obj.waveDir*pi/180) +
>                     obj.wavegauge1loc(2).*sin(obj.waveDir*pi/180)))*(1+cos(pi+pi*(i-1)/maxRampIT))/2;
> 
> Error in waveClass/waveSetup (line 270)
>                     obj.waveElevReg(rampTime, dt, maxIt);
> 
> Error in wecSim (line 118)
> waves.waveSetup(body(1).hydroData.simulation_parameters.w, body(1).hydroData.simulation_parameters.water_depth, simu.rampTime, simu.dt, simu.maxIt, simu.g, simu.endTime);
>

kmruehl, 2019-04-26:
@johnallen24 can you upload a ZIP of your run? We think the issue is that you also need to define `waves.waveSpread` as a column vector as well, but it would be helpful if we can reproduce the error to debug.

nathanmtom, 2019-05-03:
@johnallen24 Thank you for bringing this error to the development team's attention. After reviewing the error you have posted it appears you were running regular wave option in WEC-Sim so by using a waves.waveDir = [0; 45] you want to have two waves of equal amplitude and frequency to come from two directions, correct?

If the above is correct, then unfortunately WEC-Sim doesn't allow this in the regular wave option. The wave directional spreading described by Yi-Hsiang is available in the irregular wave where you will need to define the waves.Dir and the waves.waveSpread which are the same vector sizes. 

Please let us know if you desire the regular wave option to allow for two wave headings or if the irregular wave option satisfies your needs.

Cheers,
Nathan Tom

kmruehl, 2019-05-21:
@johnallen24 WEC-Sim does not currently have the ability to model regular waves with multiple incident directions, but if you would like to add this functionality to the `waveClass` and submit a pull request, we would be happy to include your addition in WEC-Sim.

closed 2019-05-21 by kmruehl