WEC-Sim/WEC-Sim #298: an issue with initializing WEC-Sim’s variant subsystems

ghleila, 2019-04-23:
I need to call on wecSim from a function but it stops running due to the following error message:

1)The variant control 'sv_b1_hydroBody' of the variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model.
2)The variant control 'sv_b1_nonHydroBody' of the variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model.

Does anyone know how I can fix this problem?
Thanks,

kmruehl, 2019-04-23:
@ghleila thanks for posting this issue to the forum. It looks like we had a similar question posted in #280, but solution wasn't found. Would you be willing to post a ZIP of your model run where you call on wecSim from a function so that we can reproduce and debug the error? 

FYI, `sv_b1_hydroBody`  and `sv_b1_nonHydroBody` are the variables that initialize the hydrobody and non-hydrobody variant subsystems in the body block of RM3/Float. The issue is likely due to the fact that the variant subsystems have not been initialized when calling on wecSim from a function. However, we will have to see the exact error in order to debug it.

ghleila, 2019-05-04:
[fmincon.zip](https://github.com/WEC-Sim/WEC-Sim/files/3144700/fmincon.zip)
Thanks for your reply. Here's a ZIP of the model we're using. In the .m file named "fmincon_ctrl", I'm trying to find optimum parameters for PTO control but it stops execution because of the error mentioned before.

kmruehl, 2019-05-07:
@ghleila Thank you for providing the ZIP of your model. I am able to reproduce your error, and am looking into a resolution.

kmruehl, 2019-05-07:
@ghleila it looks like the `wecSim.m` file you provided is an old version of the WEC-Sim executable. Please pull the latest version of `wecSim.m` from the [repository](https://github.com/WEC-Sim/WEC-Sim/blob/master/source/wecSim.m), and add the **$SOURCE** directory to your MATLAB path (http://wec-sim.github.io/WEC-Sim/getting_started.html#install-wec-sim). The WEC-Sim executable should be located in the **$SOURCE** directory, not the **$CASE** directory. Once you have the latest version of WEC-Sim source code added to your MATLAB path, please remove the `wecSim.m` file you provided, and run `fmincon_ctrl.m`. That resolved the issue for me. 

Info about WEC-Sim model files and the **$CASE** directory:
http://wec-sim.github.io/WEC-Sim/overview.html

Info about WEC-Sim source code and the **$SOURCE** directory:
http://wec-sim.github.io/WEC-Sim/code_structure.html

ghleila, 2019-05-07:
@kmruehl Thank you, problem solved.

closed 2019-05-07 by ghleila

kmruehl, 2019-05-07:
@ghleila glad that solved the issue.