WEC-Sim/WEC-Sim #299: NEMOH-BEMIO-WEC-Sim

celeparcombatir, 2019-05-06:
Dear readers,

The example of BEMIO with NEMOH is really useful, I am trying to replicate it with my own data from NEMOH for an OSWEC device.
The example works like a charm for me but if I use my own data (Nemoh.cal, Hydrostatics.dat, KH.dat, RadiationCoeffiecients.tec, ExcitationForce.tec, DiffractionForce.tec and FKForce.tec) the bemio.m gets stuck at: writing data in h5 format... with the following error as seen in the picture:
[image]

It is probably some stupid error from my part but couldn't find anything online about it, would you know what could fix this?

Kind regards,
Caspar

celeparcombatir, 2019-05-06:
I believe the value of 0.02 is wrong, this should be the amount of time steps. However if I use the default for this (1001) I get another error with indexing again.
[image]

kmruehl, 2019-05-06:
@celeparcombatir would you be willing to upload a ZIP of your NEMOH run with your BEMIO input file? That way we can debug the issue. Thank you.

celeparcombatir, 2019-05-07:
@kmruehl thank you for your fast reply. Yes of course, I have added a ZIP with all the files and my bemio.m.
[Oyster.zip](https://github.com/WEC-Sim/WEC-Sim/files/3151885/Oyster.zip)

celeparcombatir, 2019-05-09:
I believe the problem is due to the degrees of freedom of my calculations. My calculations have only 1 but I think WEC-Sim wants 6 to solve the EOM in 6 DOFs. 
I will be trying to fix this and report on it.

kmruehl, 2019-05-10:
@celeparcombatir Thanks for attaching a ZIP of your BEMIO-NEMOH run. I think your assessment is correct. WEC-Sim requires the hydrodynamic coefficients in all 6-DOF, and your NEMOH run only has 1-DOF. The error in BEMIO occurs in the for loop, iterating through the DOF `hydro.dof`. Re-running NEMOH with all 6-DOF should resolve the issue. 

```
Error in Radiation_IRF (line 39)
            ra_B = interp1(hydro.w,squeeze(hydro.B(i,j,:)),w);
```

celeparcombatir, 2019-05-14:
@kmruehl Thank you for looking into it! I had troubles with using NEMOH to make a mesh and calculate the hydrodynamic coefficients. 
I now have it working and added my Matlab file I used to make a mesh, use NEMOH and use BEMIO. 
Maybe it will be useful for somebody struggling with making a .h5 file using NEMOH for an OSWEC when they want to use WEC-Sim. 

[oswec_mesh_nemoh_bemio.zip](https://github.com/WEC-Sim/WEC-Sim/files/3179203/oswec_mesh_nemoh_bemio.zip)

closed 2019-05-14 by celeparcombatir