WEC-Sim/WEC-Sim #300: Error in h5read

vincentoneka, 2019-05-08:
Good afternoon,

I'm really annoyed because I think the error is very stupid but I need help on this kind of messages : 
[image]

I have already checked the related messages in the forum, but I can't find a solution...
I'm using Matlab R2018b
I mad a simulink model with three bodies, each one have is own .h5 file (maybe this is forbidden ?) the Matlab script stopped when reading the second body (ii=2) . The .h5 files are about 1.4 Ko
I created my own .h5 files from NEMOH and BEMIO making sure that git lfs is installed on my computer

Can you provide me the sequences to do for checking if I'm using the right version, if I have installed all the required filed, to ensure me that I'm using BEMIO correctly ?
Thank you for responding,

Best regards,
Vincent

yuyihsiang, 2019-05-21:
Hi Vincent,

Sorry for the late reply. Yes, you should only have only one h5 file instead of three. I would recommend you combine them.

I am more than happy to check to see if I can reproduce the error message if you upload your case.

Thanks,

Yi-Hsiang

kmruehl, 2019-05-30:
@vincentoneka does this resolve your issue?

closed 2019-07-15 by kmruehl