WEC-Sim/WEC-Sim #301: 'Run quiet' for WEC-Sim

glennond, 2019-05-09:
I'm calling WEC-Sim from a separate matlab optimization function, exactly as described in issue #298 and trying to log the optimization path.

1) The fmincon function has a logging option described [here,](https://www.mathworks.com/help/optim/ug/fmincon.html#busow0u-1_1) which unfortunately logs each iteration by printing it to the command window before the next function call.
Since WEC-Sim.m opens with a clc; and does a lot of it's own logging to the command window, I'm losing this information every time it is called.
 
I wanted to check if there is some sort of existing 'non-verbose' flag or argument to prevent WEC-Sim from clearing the command window and chatting so much as it runs, before I try to modify the .m myself to comment out all the clc, clear, bdclose, fprintf, toc, etc..

2) Alternatively there's another method to log fmincon iterations described [here](https://www.mathworks.com/help/optim/ug/output-functions.html), but it would rely on calling WEC-Sim from a nested function, which [matlab has problems with.](https://www.mathworks.com/help/matlab/matlab_prog/variables-in-nested-and-anonymous-functions.html) So is there any existing WEC-Sim workaround for this static workspace variable issue? 

The former option seems like the path of least resistance since the latter might require making WEC-Sim itself into a nested function.

glennond, 2019-05-10:
Solved by just using a separate 'diary' call for fmincon iteration printouts.

closed 2019-05-10 by glennond

kmruehl, 2019-05-10:
@glennond I'm glad you were able to resolve the issue, and thank you for letting us know how you resolved it. Did you also try removing the `clc` from `wecSim.m`? There is no functional requirement for `clc` in `wecSim.m`, it is only indeed to keep keep the command window clean.

crobarcro, 2019-05-10:
Yes, for the record, this is a crazy thiing to have in there by the way.

glennond, 2019-05-10:
I did try removing clc and quieting WEC-Sim,  but it ultimately wasn't necessary because I created a diary log file which was disabled while WEC-Sim was running, so the only remaining logs in that file were from fmincon.

As a side note, the second method is also possible using handle classes instead of nested functions, as described [here.](https://stackoverflow.com/questions/40003329/matlab-optimization-record-data-over-iterations) The author claims this is slower, but the performance seems acceptable to me. fmincon called WEC-Sim 90 times in about as many minutes which is similar to what I've seen using other methods.

bradling, 2019-06-12:
I second the `clc` being crazy - every time I update my version of WEC-Sim the first thing I do is remove that. 

@glennond you can also suppress output the output text by calling `[~]=evalc('wecSim');`, which is what I usually do.