WEC-Sim/WEC-Sim #302: Read_NEMOH.m - upper case Nemoh.cal

ryancoe, 2019-05-16:
On OSX and Windows, the specification of a filename for `fopen` is not case sensitive (https://www.mathworks.com/matlabcentral/answers/270711-when-matlab-read-file-file-name-is-case-insenstive). However in Linux, `fopen` is case sensitive. From what I can tell, NEMOH (and the MATLAB wrappers that go with it) use the following convention:

- `.cal` files are start with uppercase (`Nemoh.cal` and `Mesh.cal`)
- the directories are lowercase: `mesh` and `results`

This seems to be more or less flipped in [`Read_NEMOH.m`](https://github.com/WEC-Sim/WEC-Sim/blob/master/source/functions/BEMIO/Read_NEMOH.m) - the directories are uppercase and and the `.cal` files are lower case. This has never been a problem for me on OSX, but it appears to create a problem on Red Hat Linux. This was noticed by @zmorrell-sand.

dforbush2, 2019-05-21:
The most recent commit added exceptions to handle the case fopen fails to create a valid file ID for a linux operatng system within Read_NEMOH. I reviewed the other BEMIO related codes as well and did not find any similar instances of hard-coded sub-directories that could potentially cause issues, but please let me know if you encounter other problems on Linux machines

ryancoe, 2019-05-21:
Thanks Dom. From what I recall (don’t have it in front of me offsite), the directory structures created by Nemoh are always lowercase (regardless of OS). Likewise with the .cal files starting with uppercase. If this were true, you could simply change the lines for all OSes (i.e., no need for a flag). Am I misremembering?

dforbush2, 2019-05-21:
Hi Ryan,

You’re right in general. However the tutorial files in BEMIO all have manually renamed Nemoh output folders (Mesh, Results), so I wanted to make sure that Read_Nemoh could handle both Nemoh created defaults and tutorial cases.

Best,
Dom

closed 2019-05-30 by kmruehl