WEC-Sim/WEC-Sim #304: Issue with PM waves and "Traditional" freqDisc?

tedbrekken, 2019-05-23:
When using 

waves = waveClass('irregular');         % Initialize Wave Class and Specify Type
waves.H = 5;                          % Significant Wave Height [m]
waves.T = 10;                            % Peak Period [s]
waves.spectrumType = 'PM';              % Specify Wave Spectrum Type
waves.phaseSeed = 1;                    % Phase is seeded so eta is the same
waves.freqDisc = 'Traditional';

It appears all water surface elevation time series produced are the same, with waves.H having no effect.  In other words, in the code above, I always get the same water surface elevation time series, regardless of what I set waves.H to.

nathanmtom, 2019-05-24:
@tedbrekken You are correct, the Pierson-Moskowitz (PM) spectrum is only a one parameter spectrum in WEC-Sim which is defined by the peak period. We had previously used the \alpha_{pm} term as a coefficient that we could tune to match a given significant wave height; however, we found that this result is nearly identical to the solution to the Bretschneider (BS) spectrum. Therefore, we chose to keep the Pierson-Moskowitz (PM) spectrum as a one parameter spectrum that does not use the significant wave height in the construction of the spectral shape. I'd suggest using the Bretschneider (BS) spectrum if you'd like to match both the peak period and significant wave height. For more information, please refer to the wave spectrum theory section in the WEC-Sim documentation [here](https://wec-sim.github.io/WEC-Sim/theory.html#pierson-moskowitz-pm).

Please let us know if this has answered your question.

Cheers,
Nathan Tom

closed 2019-05-30 by kmruehl

hossein-mehdipour, 2021-07-08:
Hello @nathanmtom I wanted to ask you 2 questions about this topic if you don't mind:
1. I saw that the "Bretschneider" spectrum is not included in [this link](https://wec-sim.github.io/WEC-Sim/man/api.html?highlight=spectrum#objects.waveClass.spectrumDataFile) under the `spectrumType `section, I was wondering if there is a reason it's not included there (and is included in your old manual)?
2. When using for example BS spectrum, what should I use for the `Waves.H` (the significant wave height), should I use the **Hs** or **Hm0** or even **H100**?

nathanmtom, 2021-07-08:
@hossein-mehdipour Thank you for your question and I'll do my best to answer. 

1. The WEC-Sim team tries to follow the [IEC TC114 - Marine energy - Wave, tidal and other water current converters standards](https://www.iec.ch/dyn/www/f?p=103:7:711304648759346::::FSP_ORG_ID,FSP_LANG_ID:1316,25) as best as possible. In the [IEC TS 62600-2:2019](https://www.iec.ch/dyn/www/f?p=103:22:711304648759346::::FSP_ORG_ID,FSP_LANG_ID:1316,25) standard the spectrum analytical formulation used within WEC-Sim is defined as Pierson-Moskowitz rather than Bretschneider. There is no difference between the expression and the change is just with the labelling.
2. When using WEC-Sim the `waves.H` function refers to the significant wave height which is fairly common statistic measure provided for a sea state. I'll need you to define your variables before providing comments so as not to mistake your notation for others we are familiar with.

I hope this helps answer your questions. 

Cheers, 
Nathan
National Renewable Energy Laboratory

hossein-mehdipour, 2021-07-09:
@nathanmtom 
1. Thank you for the clarification.
2. I am adding the definitions below with screenshots.

[image]

and
[image]

nathanmtom, 2021-07-09:
@hossein-mehdipour Thank you for starting to list the variables you referenced in your previous comment. In your attachment it appears you have assumed a Rayleigh Distribution while most of the wave spectrums are assumed to follow a Gaussian Distribution which changes the relationship between the RMS of the sea surface and the significant wave height, see [here](http://falk.ucsd.edu/pdf/Lectures6_7.pdf). The `H_{m0}` in WEC-Sim corresponds to the significant wave height with many other references utilizing `H_{s}`. The `_{m0}` subscript corresponds to the 0th order moment of the wave spectrum and I refer you to the following example for more information, see [here](https://ocw.mit.edu/courses/mechanical-engineering/2-22-design-principles-for-ocean-vehicles-13-42-spring-2005/readings/lec6_wavespectra.pdf). 

I hope this information provides additional information to help answer your questions, but let us know if you still have a question or the above is not clear. 

Cheers, 
Nathan
National Renewable Energy Laboratory