WEC-Sim/WEC-Sim #306: Error in reading Aqwa Output file

AmirH7076, 2019-07-09:
Hello
Thanks to WEC-Sim
I want to simulate a simple Two-body Point Absorber in  WEC-Sim . Aqwa's .AH1 file includes the specifications of each part. But the Read_AQWA  m-file only process the hydrodynamic data of one part. Actually the number of body is set 1 not 2. I found the problem and  then I modified the .AH1 file manually. The problem was in first line of .AH1 file:
 * DATE .....
  1  5  99    -180.00  .....

  2  5  99     -180.00  ....

is modified to:
* DATE ....
  2  5  99  -180.00 ....

 but bellow error occurred:

Unable to perform assignment because the size of the left side is 1-by-6 and the size of the
right side is 0-by-0.

Error in Read_AQWA (line 121)
                            hydro(F).A((ind(2)-1)*6+i,((ind(1)-1)*6+1):(ind(1)*6),ind(3)) =
                            tmp;

Error in bemio (line 4)
hydro = Read_AQWA(hydro,'ANALYSIS.AH1','ANALYSIS.LIS');

Please help me.
best regards.

kmruehl, 2019-07-15:
This is related to issue #295

dforbush2, 2019-07-15:
Hello,

Could you please provide your .AH1 and .LIS file? From your description of the error, it appears that line 121 is attempting to read a line of the AH1 file that does not exist, or is empty. The location of Body #2 data (or higher) is assumed based on the number of frequencies and wave headings in line 118, as defined in the first line of the AH1 file you identified:

[118] - tmp = str2num(raw1{ln+(m-1)*hydro(F).Nb*hydro(F).Nf*6+(k-1)*hydro(F).Nf*6+(j-1)*6+i});

As a quick fix, you can manually change this line to select the location in your AH1 file where the appropriate data is located, perhaps most easily by replacing "ln" with this line number.

Your error could be explained if your AH1 file (for whatever reason) does not actually contain data for both bodies, resulting in a 0 x 0 RHS. However, I would very much appreciate if you could provide your AH1 file so that I can ensure this is not a compatibility issue with this version of AQWA

Thank you

dforbush2, 2019-07-16:
Hello again,

Without seeing your specific files, I am unsure if this will solve your issue, but a compatibility issue was found between Read_AQWA and new versions of AQWA. This has since been fixed. Please update Read_AQWA.m and try again.

Thank you.

kmruehl, 2019-07-30:
This was resolved by commit: https://github.com/WEC-Sim/WEC-Sim/commit/49823657e056b6baa7239e44dd3782dedae734b9

closed 2019-07-30 by kmruehl