WEC-Sim/WEC-Sim #307: What forces would be seen on load cell?

crobarcro, 2019-07-20:
Hello,

I'm hoping I can get some help answering a validation question. To take the RM3 as an example, if I were to place a load cell between the float and it's connection point to the PTO, what force would I see on the load cell In terms of a sum of the forces calculated in WEC-Sim? 

This is relevant for validation of forces using a load cell like this. It is confusing due to the added mass force and how this is handled in WEC-Sim (it is post-prcessed after the simulation to something quite different to what is seen during the simulation).

Thanks!

yuyihsiang, 2019-07-30:
Hi

What I would typically do is add a non-hydrobody at the location, in between the float and your PTO, where you have the load cell and use the output from it. This would avoid the added mass issue.

Yi-Hsiang

kmruehl, 2019-07-30:
If you want to output the load from the load cell at the location of the pto/constraint you can also call on the following variable from the output data structure, for example `output.constraints.forceConstraint` 

http://wec-sim.github.io/WEC-Sim/code_structure.html#id2

kmruehl, 2019-08-06:
@crobarcro does this resolve your issue?

crobarcro, 2019-08-06:
Thank you, yes, it does.

closed 2019-08-06 by crobarcro