WEC-Sim/WEC-Sim #308: Added mass force calculation

crobarcro, 2019-07-23:
I have a few questions about the added mass force calculation in WEC-Sim

In WEC-Sim, the added mass force is calculated by taking the sum of the diagonal terms of the added mass matrix, multiplying by some factor, and then adding this to the body mass like so:
```
tmp.adjmass = sum(tmp.fadm(1:3))*adjMassWeightFun;
obj.mass = obj.mass + tmp.adjmass;
```
`tmp.adjmass` is then also subtracted from the every diagonal term of the added mass matrix like so:

```
obj.hydroForce.fAddedMass(1,1) = obj.hydroForce.fAddedMass(1,1) - tmp.adjmass;
obj.hydroForce.fAddedMass(2,2) = obj.hydroForce.fAddedMass(2,2) - tmp.adjmass;
obj.hydroForce.fAddedMass(3,3) = obj.hydroForce.fAddedMass(3,3) - tmp.adjmass;
```
So my questions are: 

1. I understand the general rationale here (which is explained in https://github.com/WEC-Sim/WEC-Sim/issues/161#issue-206825599), but why sum the masses, rather than, say, take the largest value?

2. The default value for `adjMassWeightFun` is 2, how was this value chosen, and in general, how should one choose a value for this?

Thanks!

yuyihsiang, 2019-07-30:
1. The largest value can be tricky. If you happen to have the largest added-mass in surge but you have a simulation that is only in heave, this added-mass may cause an issue. The factor gives the users a way to adjust the value if they want.
2. The value was selected based on past experience. You are more than welcome to change it.

crobarcro, 2019-08-02:
When you say past experience, do you mean past experience that it makes the simulation more stable, or past experience that it produces more accurate results?

yuyihsiang, 2019-08-07:
Typically, the value would not affect the accuracy of the simulation as long as the simulation is stable.

crobarcro, 2019-08-16:
hmmm, ok, it does kind of raise the question of why bother calculating added mass at all then, I mean, if you double it, it must have some kind of effect on the simulation.

crobarcro, 2019-09-03:
This makes sense, I see now how you are doing this. Thanks for the explanation.

closed 2019-09-05 by yuyihsiang