WEC-Sim/WEC-Sim #312: Possible bug in writing control surface values?

crobarcro, 2019-07-24:
Write_H5 contains the following lines:

```
    if isfield(hydro,'md_mc')==1 % Only if mean drift variables (momentum conservation) have been calculated in BEM
        H5_Create_Write_Att(filename,['/body' num2str(i) '/hydro_coeffs/mean_drift/momentum_conservation/val/'],permute(hydro.md_mc((n+1):(n+m),:,:),[3 2 1]),'Value of mean drift force (momentum conservation)','');
    end
    if isfield(hydro,'md_cs')==1 % Only if mean drift variables (control surface approach) have been calculated in BEM
        H5_Create_Write_Att(filename,['/body' num2str(i) '/hydro_coeffs/mean_drift/control_surface/val/'],permute(hydro.md_mc((n+1):(n+m),:,:),[3 2 1]),'Value of mean drift force (control surface)','');
    end
```
In the first `if` you check for the `md_mc` field, then use it, however, in the second `if`, you check for `md_cs`, then don't use it, but instead have the exact same line from the previous `if`.
I assume the statement in the second `if` was meant to address `hydro.md_cs` not `hydro.md_mc` again?

yuyihsiang, 2019-07-30:
Fixed and pushed

Thanks,

Yi-Hsiang

kmruehl, 2019-07-30:
resolved by commit: https://github.com/WEC-Sim/WEC-Sim/commit/29f4cc29cf785b75837dffd2fd818494d452efc4

closed 2019-07-30 by kmruehl