WEC-Sim/WEC-Sim #315: I can't see the arrow to go to the sub-blocks in the RM3 file.

GyeongKim, 2019-07-29:
Hi.
I am a very beginner for the WEC-Sim and a user of MATLAB 2019a.
Now I am studying how to use the WEC-Sim, but I can't see the arrow to go to the sub-blocks in the RM3 file.
Could somebody let me know how to do it?
Many thanks for your help.
[image]

kmruehl, 2019-07-30:
It looks like the arrow button has been removed. You can either hit `Ctrl+U`, or you can right click on the block and select Mask->Look Under Mask. 

[image]

closed 2019-07-30 by kmruehl

kmruehl, 2019-07-31:
No problem, glad to help.