WEC-Sim/WEC-Sim #316: Jenkins Test Issue

zmorrell-sand, 2019-07-30:
I am opening this issue to test Jenkins CI using GitHub WebHooks.  If this successfully makes Jenkins build, I will close this issue.  Otherwise, I will type single character responses to illicit another test.  The issue will be closed when Jenkins is functional.

zmorrell-sand, 2019-07-30:
a

zmorrell-sand, 2019-11-04:
a

zmorrell-sand, 2020-01-13:
Jenkins is now set up to poll the repository every hour to rebuild.  At the time of writing for this comment, the test cases are deprecated and do not run, but Jenkins is operable.  When the tests are fixed, I will have Jenkins begin sending failure notifications to @kmruehl

kmruehl, 2020-01-15:
@zmorrell-sand I'll work on updating the test cases.

kmruehl, 2020-01-15:
I updated the verification file

closed 2020-01-15 by kmruehl