WEC-Sim/WEC-Sim #317: Is it possible to define a spring connection and a constraint with the expected restraint degree of freedom

Tudore, 2019-08-01:
Is it possible to define a spring connection and a constraint with the expected restraint degree of freedom?
1. Could we define a spring (linear elongation or nonlinear) in SIMULINK model?
2. Now there are some pre-defined constraints. Are we able to create a 'customized' constraints? For example to restrain x,y,z and rz. This is to simulate a body to be restrained by mooring dolphin.

Thanks!
Regards,
Tudore_Zhang

kmruehl, 2019-08-02:
@Tudore 
1. I've never done this before, but you should be able to define a non-linear spring in Simulink. It looks like this can be done using the [Nonlinear Translational Spring block](https://www.mathworks.com/help/physmod/sdl/ref/nonlineartranslationalspring.html), or a similar spring block provided by Simulink.
2. Yes, you can define custom constraints using the [Simscape Multibody Joints and Constraints](https://www.mathworks.com/help/physmod/smlink/ref/-inventor-constraint-simmechanics-joint-correspondence.html) library. I recommend modifying one of the existing WEC-Sim PTO/Constraint library blocks to build you custom block.

kmruehl, 2019-08-07:
@Tudore did this resolve your issue?

Tudore, 2019-08-13:
> @Tudore did this resolve your issue?

Thanks, Kelly. I need to explore this function. Will give feedback once it is done.