WEC-Sim/WEC-Sim #319: In-simulation access to current wave height

glennond, 2019-08-16:
Hello WEC-Sim community,
I'd like to design a PTO control system that responds to the current wave conditions in-simulation. For example, using a rolling average period over a shorter window, to allow for longer simulations that undergo changes in the primary wave period throughout the simulation.

Is there a way to access an instantaneous waveheight at a body? As opposed to waves.T and waves.H, I'd like the control to respond on a shorter timescale. 

I'd also like to avoid having to use an etaImport and be able to respond to any of the irregular spectra.

thanks,
-David

glennond, 2019-08-18:
I seem to have found a complicated and cumbersome way to do this
[image]

Here is the text of the function:
`function eta = waveElevNow(curSimTime, wavesMat)

eta = wavesMat((find(wavesMat(:,1)==curSimTime)),2);`

**Please still let me know if there is a simpler way I'm missing.**

yuyihsiang, 2019-09-05:
Here is the FFT function. Hopefully, that can be helpful for you.

[Archive.zip](https://github.com/WEC-Sim/WEC-Sim/files/3579945/Archive.zip)

Yi-Hsiang

kmruehl, 2019-11-05:
@glennond did this resolve your issue?

glennond, 2019-11-07:
Yes, I no longer have an issue here

closed 2019-11-08 by kmruehl