WEC-Sim/WEC-Sim #320: Current direction issue

jackhughes1, 2019-10-02:
Hi, I'm trying to use the new currents implementation in waveclass. 

When setting the current direction the simulation performs as expected, however if I rotate the body into flow at an angle that should produce the same result I get unexpected, unstable results. 

This image shows the forces in surge and sway of a hemispherical geometry (taken from the applications repository) with C_D = [1 1 1] and a current speed of 2 m/s applied at a current direction of 45 degrees. 
[image]

The following results show the same geometry, orientated 45 degrees to flow of 2 m/s and current direction of 0 degrees. 
[image]
[image]

I know of the small angle assumption, but the rotation matrix in the Morison equation implementation appears to get around this - however the results do not show this. 

Is there any reason for these results and is there a possible way to achieve reasonable results when orientating a geometry to the flow rather than vice versa. 

Thanks,
Jack

nathanmtom, 2019-10-11:
@jackhughes1 Thank you for bringing this issue to the development teams attention. 

I have a few questions to ask to clarify the conditions you are running:

1. If I understood you correctly, you are using the files from the IEA_OES_Task10_freeDecay folder in the WEC-Sim Applications in the WEC-Sim repository.
2. You have stated that you are using the C_{D} = [1 1 1], but are you using the other default values? I'm trying to reproduce your first plot above, but it appears you might have changed some values. Perhaps you could share your wecSimInputFile.m? Also, the .slx file if you are moving in both X and Y (surge and sway) then you may have changed the constraint. 
3. When you state the orientation was set at 45 degrees, you are referring to the yaw angle, and the following options were used in WEC-Sim: 1) body(1).initDisp.initAngularDispAxis    = [0 0 1];
body(1).initDisp.initAngularDispAngle   = 45.0/180*\pi?

We generally have assumed that the body would be oriented with 0 heading and the BEM input data would be calculated with a vary wave heading and the device response can be calculated. However, currently WEC-Sim doesn't update the calculation of the the wave diffraction forces with the instantaneous yaw angle; however, this is currently under development. Therefore, you are correct that the two cases you have proposed should provide identical results, but we suggest keeping the body orientation at 0 and moving the current direction. The development team will update once we have new information to share.

Please let us know if you need any clarification from us or you have additional questions.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

kmruehl, 2019-11-12:
@jackhughes1 has this resolved your issue?

closed 2020-05-05 by kmruehl