WEC-Sim/WEC-Sim #321: MOORDYN ISSUE - NaN DETECTION

sricharanvvs, 2019-10-17:
Hi ,
I have gone through the related posts but unable to resolve. Mine is a floating WEC, have used MoorDyn. Untiill I attach or include Moordyn the simulation is running well. 
Also, Time step is 0.01, ramp time also kept small. (2019b)
-- One thing I dont understand is the locations of PTO, Mooring etc. With which reference the location is to  be entered?
-- lines.txt is also included in mooring folder: Simulation says, NaN value is generated.
-- The below error pops up irrespective of values I change in lines.txt
Error using wecSim (line 261)
Derivative of state 'FWEC3_CBHEAVE.Translational.CONSTRAINT_Prismatic_Joint.Pz.v' in block
'FWEC3_CBHEAVE/Translational/CONSTRAINT (Prismatic Joint)' at time 0.01 is not finite. The simulation will be stopped. There may
be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening
the error tolerances)
I am attaching my files. Please review and help me through. 
Thanks, 
Regards,
Charan - IITM
[image]

_Originally posted by @sricharanvvs in https://github.com/WEC-Sim/WEC-Sim/issues/205#issuecomment-542574143_

yuyihsiang, 2019-11-05:
Hi Charan,

The reference location should be [0 0 0] for the mooring since you have defined everything assuming an origin of [0 0 0] in moorDyn.

I noticed that your mooring line is 1.3 m but your vertical distance is 2.4m. I assume you are trying to have an pre-tension in the mooring line. However, based on the error message that moorDyn is crashing when initializing the system. 

You may want to change the SOLVER OPTIONS for the moorDyn, such as the time-step size and others for moorDyn. The other way around is to scale everything back to full scale.

Best,

Yi-Hsiang

sricharanvvs, 2019-11-10:
Hey, Thanks a lot for the response!
Its working now with the below lines.txt file.
1. X, Y values of the lines are on the edges of my body but the 'Z' value I have given from the CG of that body which is -1.08 from the MSL. So -0.2 is from the CG (-1.08): is it right ? I mean the mutual references are right?
2. I am unable to see the cables physically either in the simu.explorer nor in the paraview. I have set Flags/Outputs as 'p' but still the same? Am I missing anything?

- Regards,
Charan, IIT M 
[lines.txt](https://github.com/WEC-Sim/WEC-Sim/files/3828430/lines.txt)

yuyihsiang, 2019-11-26:
1. -0.2 is from the reference point

2. simu.explorer does not include the mooring line. Regarding Paraview, you should be able to see the mooring line. You may want to check if you find additional mooring output files.

Yi-Hsiang

sricharanvvs, 2019-12-03:
Thank you for the response. 
My other query:
1.I am working on multibody WEC where in Aqwa the effect of  body interactions was considered while calculating hydrodynamic parameters. Should I still include body to body interaction 'b2b' line in the input file?
I have tried giving it but showed an error. 
I have 9 bodies in total.. how should I use this 'b2b' in this case if I need to include it?

Regards
Charan

kmruehl, 2019-12-18:
@sricharanvvs we would _not_ recommend turning the b2b option for 9 bodies.

closed 2019-12-18 by kmruehl