WEC-Sim/WEC-Sim #322: Suspictious indexing in Read_NEMOH.m

crobarcro, 2019-11-05:
Read_NEMOH.m has the following around line 289:

```
for nn =1:3
    M(nn,nn) = hydro(F).Vo(m)*hydro(F).rho;
end
```
`Vo` is indexed using the variable `m`, which was last set by a loop around line 109, and is therefore probably always equal to hydro.Nb (the number of hydrobodies). Is this the intention?

crobarcro, 2019-11-05:
I would also like to suggest that using meaningful variable names would be helpful in understanding the code.

crobarcro, 2019-11-05:
Same goes for use of `m` here:

```
    %% Read Inertia
    if hydro(F).Nb == 1
        fileID = fopen(fullfile(filedir,'Mesh','Inertia_hull.dat'));
        if linuxFlag && fileID==-1
            fileID = fopen(fullfile(filedir,'mesh','Inertia_hull.dat'));
        end
    else
        fileID = fopen([fullfile(filedir,'Mesh','Inertia_'),num2str(m-1),'.dat']);
        if linuxFlag && fileID==-1
            fileID = fopen([fullfile(filedir,'mesh','Inertia_'),num2str(m-1),'.dat']);
        end
    end
```

dforbush2, 2019-11-13:
Hello,

I have reviewed the indicated sections and you are correct. It appears that the loop beginning on line 109 (that loops through the number of bodies) is repeated in a few other places wherever calculations are specific to particular bodies, but is not properly included in the "reading Kochin files" section.

I'll take a look at this code and push an update once corrected.

I also agree with your comment regarding intuitive variable names. We have attempted to follow this convention with variables and structures, but have not been so rigorous in this regard with temporary variables like indices.

Thank you

crobarcro, 2019-11-13:
Yes I suggest something really obvious like `bodyind` for bodies etc, while verbose, makes spotting errors like this much easier.

dforbush2, 2019-12-04:
Read_Nemoh has been updated so that the Kochin function handles multiple bodies and wave headings.

kmruehl, 2019-12-18:
@crobarcro did this update resolve your issue?

closed 2019-12-18 by kmruehl