WEC-Sim/WEC-Sim #324: WEC-sim validation

frrfrz, 2019-11-11:
Hello, i am trying to do a validation of WEC-sim (for my master thesis) using the RM3 model and the data obtain from the following papers:
-1-Experimental Wave Tank Test for Reference Model 3 Floating-Point Absorber Wave Energy Converter Project (Y.H. Yu, M. Lawson et al.). TEST #2 (just heave).
-2-Preliminary Verification and Validation of WEC-Sim, an Open-Source Wave Energy Converter Design Tool (K.Ruehl, C. Michelen et al.).

1) I can't undertand if the experimental data of 2 are the same of 1, and if they are releted, how they are related?
2)in order to make a validation with the data from 1, i think that i should insert wave period, wave height and pto damping coeff. (table 6) inside the wecsiminputfile, moreover i should insert a 1DOF constraint in the simulink model, is that correct?
What about the simulation time, ramp, time step and viscous drag? Shall i change something else?
3)In order to obtain a set of data to compare with the measured relative motion (figure6) i think to extract the relative motion from output.ptos.position and calculate the RMS and divide it by the wave height, is this correct?

Thanks in advance for your help.

kmruehl, 2019-11-12:
@frrfrz, thanks for you question regarding the RM3 experimental data used for WEC-Sim v1.0 validation. I've responded to a few of your questions **in line below,** and @yuyihsiang will address the remainder of your questions. 

1.	I can't understand if the experimental data of [2] are the same as [1], and if they are related, how they are related?
**The experimental data in [2] is based on the RM3 design that was tested at a 1:33 scale at Scripps Institute of Oceanography in San Diego in December 2011. This corresponds to the experimental data in [1] listed as Test Number 3 (listed in Table 1).** 

2.	In order to make a validation with the data from 1, i think that i should insert wave period, wave height and pto damping coeff. (from table 6) inside the wecsiminputfile, moreover I should insert a 1DOF constraint in the simulink model, is that correct?
**If you are trying to reproduce [2], you should use Table 9 from [1]. But yes, you should use a Translational PTO between the float and spar, as shown in Figure 1 from [2].**
What about the simulation time, ramp, time step and viscous drag? Shall I change something else?
**[2] states, “The simulations of the RM3 WEC were performed at full scale for wave height, H = 3 m, and for wave periods ranging from 8 to 18 s, with a time-step of 0.01 s. For the WEC-Sim simulations, the PTO damping was assumed to be linear, and set to the average full-scale experimental damping value, 2,500 kN-s/m. Viscous drag plays a significant role in WEC motion, and tuning its numerical value is critical, especially for comparison to small-scale experimental data. In these simulations, the WEC-Sim code was run with the float drag coefficient set to 1.4, and the damping plate set to 4.3.”**

3.    In order to obtain a set of data to compare with the measured relative motion showed in 1 (figure6) i think to extract the relative motion at every instant from output.ptos.position and calculate the RMS and divide it by the wave height, is this correct?
**I believe you should be using Figure 16 if you want to compare to [2].** 

[1] [RM3_experiments_62951.pdf](https://github.com/WEC-Sim/WEC-Sim/files/3838646/RM3_experiments_62951.pdf)
[2] [2014 - OMAE 2014-24312_WEC-Sim_Ruehl.pdf](https://github.com/WEC-Sim/WEC-Sim/files/3838645/2014.-.OMAE.2014-24312_WEC-Sim_Ruehl.pdf)

frrfrz, 2019-11-13:
Thanks for the fast response, so to make a validation i should take T,H and PTO damping coefficient from [1](table9) and the drag coefficients from [2], right?
-Do you think is also needed to model the mooring lines with MoorDyn?
-Can i use the .h5 file of the RM3 tutorial or Should i create a new .h5 file using a water depth=49.5?
-After the wecsim simulation is run i take the relative motion from output.ptos.position but i am not sure about how is calculated the RMS relative motion in [1](table 9).

yuyihsiang, 2019-11-24:
We also have done the validation study again in 2018. You can find details in 
Yu Y.-H., Tom N., and Jenne D., 2018, “Numerical Analysis on Hydraulic Power Take-Off for Wave Energy Converter and Power Smoothing Methods,” 37th International Conference on Ocean, Offshore and Arctic Engineering, OMAE, Madrid, Span.

-Do you think it is also needed to model the mooring lines with MoorDyn?
You can also use the mooring stiffness matrix instead MoorDyn

-Can i use the .h5 file of the RM3 tutorial or Should i create a new .h5 file using a water depth=49.5?
You can, but I would recommend you rerun WAMIT with the correct water depth. MORE IMPORTANTLY, the moment of inertia for the spar/plate in the tutorial is different from the one described in the papers and experimental report.

-After the wecsim simulation is run i take the relative motion from output.ptos.position but i am not sure about how is calculated the RMS relative motion in [1](table 9).
Table 9 is just averaged RAO. For WEC-Sim, you can divide the output.ptos.position with wave amplitude and find the peaks. It should be relatively constant in WEC-Sim simulations.

frrfrz, 2020-01-28:
Dear Dr. Yu, I am not sure about how to build the mooring matrix using the data from [1] test #3. 
Can you please provide me some references to understand that or, if you have it, the wecsim input file for that test? In this way i could see the mooring matrix and trying to understand how is built.
I want to run my own test on wecsim in order to obtain the results of test #3 and insert this validation of wecsim on my mater thesis.
Thanks for your time and your precious help.

Fabrizio

yuyihsiang, 2020-02-09:
To answer your question:
1. It makes sense to me.

2. Please check the example on http://wec-sim.github.io/WEC-Sim/advanced_features.html#viscous-damping-and-morison-elements

3. Both 1-DOF and 3DOF. You can also see. An updated validation study was presented at
Yu, Y.-H., Tom, N., & Jenne, D. (2018). Numerical Analysis on Hydraulic Power Take-Off for Wave Energy Converter and Power Smoothing Methods. 37th International Conference on Ocean, Offshore and Arctic Engineering.

4. The mooring stiffness in the table was for each mooring line. There were four lines, as shown in Fig. 12. YOu will need to convert it back to full scale and multiple by 4 since we have 4 lines. 

5.  Yes,

Yi-Hsiang

frrfrz, 2020-02-10:
Dear Dr Yu, 

Thank you so much for your reply and your patience. Now i should be able to proceed with my calculations.

Fabrizio

closed 2020-02-18 by kmruehl

ckberinger, 2021-07-13:
Hi all,

Can you expand on why you multiply the mooring stiffness by 4 if you have 4 mooring lines? The mooring lines are likely in opposing directions. And in reality, when 2 lines are in tension, the other 2 will likely not be in tension. What is the justification for multiplying by 4?

Courtney