WEC-Sim/WEC-Sim #326: Tutorial using Nemoh in Matlab

ammar-alfaili, 2019-11-27:
Hello

I want to simulate wave forces on a cylinder in Nemoh. I wanted to see if anyone can guide me since it is the first time i am doing this, implementing/using  Nemoh in Matlab. 

thank you in advance

yuyihsiang, 2019-11-27:
See the example from Nemoh Website
https://lheea.ec-nantes.fr/logiciels-et-brevets/nemoh-examples-192931.kjsp?RH=1489593406974

You should be able to modify the example and may answer your questions.

Yi-Hsiang

ammar-alfaili, 2019-11-27:
yes, I have seen that example but I still don't know how to use Nemoh in matlab. That's the main problem. 

\Ammar

yuyihsiang, 2019-12-18:
Unfortunately, we only provide an interface with Nemoh, and this is a question for Nemoh.

cadu55br, 2020-01-02:
I created a MATLAB script for a Vertical Cylinder with Flat bottom (VCF_Tutorial_Arnal_axiMesh_Nemoh_Bemio.m), based on the presentation "NEMOH: a tutorial using the Matlab wrapper", by Vincent Arnal, 11-2014 (arnal.vinc@gmail.com). See attached. 
I hope it can help you. Good luck!
Carlos

[Example of MATLAB script for running NEMOH.txt](https://github.com/WEC-Sim/WEC-Sim/files/4017019/Example.of.MATLAB.script.for.running.NEMOH.txt)

[NEMOH Tutorial, by V.Arnal.pdf](https://github.com/WEC-Sim/WEC-Sim/files/4017020/NEMOH.Tutorial.by.V.Arnal.pdf)

kmruehl, 2020-01-03:
@cadu55br thank you for providing this example.

closed 2020-01-03 by kmruehl

ammar-alfaili, 2020-01-06:
Hello Carlos

Thanx a lot. This was very helpful. I appreciate it.

Best regards

Ammar

dith96, 2023-02-23:
Hello..
I just found this page while looking for a tutorial on installing and using Nemoh.. 
I want to simulate waves for OWC in Nemoh for my final project, but until now I'm still confused about how to use it..
Is there anyone can guide me using Nemoh? Please.. help me anyone.. This is my first time trying  Nemoh.

Thank you..

Javeria503, 2024-10-18:
hello i am a beginner to NEMOH i did not have any idea about that how to used it if someone can help me with this it will be much helpful for me.

kmruehl, 2024-10-18:
@Javeria503 unfortunately we do not provide NEMOH support. However, we do have some BEMIO examples that using NEMOH here: https://github.com/WEC-Sim/WEC-Sim/tree/main/examples/BEMIO/NEMOH

cadu55br, 2024-10-22:
@Javeria503, in order to try to help you, I am sending below the link to a PDF copy of my master's dissertation, presented to COPPE/UFRJ on 21/December/2020, as a partial fulfillment of the requirements for the degree of Master of Science (M.Sc.):
"STUDY ON THE MAXIMIZATION OF THE ABSORBED POWER OF POINT ABSORBER WAVE ENERGY CONVERTERS IN RIO DE JANEIRO, BRAZIL"
"ESTUDO SOBRE A MAXIMIZAÇÃO DA POTÊNCIA ABSORVIDA DE CONVERSORES DE ENERGIA DAS ONDAS DO TIPO ABSORVEDOR PONTUAL NO RIO DE JANEIRO, BRASIL"

Link: 
https://lnkd.in/dX2g_e3H

Unfortunately, the text is in Portuguese, but with an online translator it is possible to understand most of the text.

I suggest you pay attention to APÊNDICE D (Appendix D, page 169/246), which contains "Examples of Inputs and Outputs of the Programs Used", as well as the respective MATLAB scripts.

I hope this information is useful to you.
Good luck and best regards.
Carlos

kmruehl, 2024-10-22:
@cadu55br thank you!

Javeria503, 2024-11-08:
@cadu55br Hello can you guide me how to use BEM for only diffraction problem. I look your examples but in my case the structure is fixed not floating. Thanks _

bibble18, 2025-06-20:
@cadu55br Hi carlos your files have been very helpful to the project I am currently working on. However I am facing issue with running Nemoh and BEMIO for the sphere for now(it worked well on cylinder), is there any way I can have your contact(email or anything) that would be very helpful.