WEC-Sim/WEC-Sim #327: Array WECs model establishment

RaymondValentine, 2019-11-29:
Hello, I am studying array wave energy device analysis from the following paper：
[1] Mankle, Hannah, Yi-Hsiang Yu, and Bryony DuPont. WEC-Sim Array Development and Experimental Comparison Study. No. NREL/CP-5000-74083. National Renewable Energy Lab.(NREL), Golden, CO (United States), 2019.

I would like to ask if additional damping and joint constraints need to be added to each float when calculating in BEM (e.g. AQWA), just as PTO damping and joint constraints sometimes have a large impact on waves?

Thank you in advance.

yuyihsiang, 2019-12-19:
I believe we first tuned the model for a single device with linear damping and you can find details in
Tom, N., Ruehl, K., & Ferri, F. (2018). Numerical Model Development and Validation for the WECCCOMP. 37th International Conference on Ocean, Offshore and Arctic Engineering. Madrid, Span: OMAE.

Then, we use the model to run the array case.

Yi-Hsiang

yuyihsiang, 2020-01-15:
Hope that answers your question. Let us know if you need us to reopen this one.

closed 2020-01-15 by yuyihsiang