WEC-Sim/WEC-Sim #328: dof bemio wamit

avidortelem, 2019-12-17:
Hello,

I am new to wecSim and I try to use bemio for my wamit files.

The Read_WAMIT function runs without error but after I have an error on the function Radiation_IRF (row 39).

Indeed this function try to read the damping coefficient B on every DOF as can be seen in the folllowing code extract:
(begin of extract)
for i = 1:sum(hydro.dof);
        for j = 1:sum(hydro.dof);
            ra_B = interp1(hydro.w,squeeze(hydro.B(i,j,:)),w);
(end of extract)
with hydro.dof=6 given by Read_WAMIT

But my wamit files as been with only surge and the dimension of B given by Read_WAMIT is (1,1,Nf) .... 

It looks like the function Read_WAMIT does not care of the actual dof (it does look into the .pot file of WAMIT where the dof are written) and put hydro.dof=6 by default. Same for the Radiation_IRF which do a loop from 1 to sum(hydro.dof).

So I think, whatever my wamit files is only surge or not, that the dimension of B given by Read_WAMIT should always be (6,6,Nf) for one body... Is there an issue with my wamit files that could explain that it is not the case?

Thanks for you help,
Best Regards

avidortelem, 2019-12-18:
I actually have the same issue with Read_Nemoh

Br

kmruehl, 2019-12-19:
@avidortelem since WEC-Sim is a 6DOF solver we require the BEM solution to be 6x6xNf. If you are only interested in the surge response of a device, we recommend running the BEM solution in 6DOF, but then constraining the device's response to surge motion only in WEC-Sim. This can be done in WEC-Sim using `Constraints`, read here for more info: http://wec-sim.github.io/WEC-Sim/code_structure.html#constraint-class/  and http://wec-sim.github.io/WEC-Sim/code_structure.html#constraints

closed 2019-12-19 by kmruehl