WEC-Sim/WEC-Sim #330: Nemoh results from tutorial\bemio\rm3

frrfrz, 2020-01-10:
Hello, i am trying to replicate the the same test using nemoh, but i am not able to get the same results, running mesh.exe i obtain the same hydrostatics and kh results but when i run preprocessor.exe, solver.exe and postprocessor.exe i obtain some completely different results. I suppose that the problem could be my mesh, are there any ''rules'' for creating the nemoh mesh?
I create the mesh with rhino using the command mesh from surface.
In the attachment you will find my mesh and the nemoh folder where i run the code and the results as well.
Thanks in advance for your help.
[nemoh.zip](https://github.com/WEC-Sim/WEC-Sim/files/4046789/nemoh.zip)

yuyihsiang, 2020-01-15:
I was trying to run you BEMIO to check it, but it can not find the hydorstatic.dat file.

Yi-Hsiang

frrfrz, 2020-01-15:
I am sorry i forgot to attach that file, now in the attachments you will find the folder ready to run with bemio. Thank you very much.
[nemoh.zip](https://github.com/WEC-Sim/WEC-Sim/files/4066894/nemoh.zip)

DavidTran2008, 2020-01-16:
Hi frrfrz,

As the official NEMOH release has not been implemented irregular frequency removal functionality, your issue is likely caused by this function rather than mesh resolution. However, higher mesh resolution could somewhat avoid irregular frequency issue as well. You can take a look at the attached file for details.

Regarding irregular frequency removal functionality, you may want to try it using OpenWARP code.
[NEMOH Issue330.pdf](https://github.com/WEC-Sim/WEC-Sim/files/4073845/NEMOH.Issue330.pdf)

Best,
Toan

frrfrz, 2020-01-16:
Hello, Toan, Thank you so much for the fast response and you time, i am going to make one last attempt using an higher mesh resolution, and if this way doesn't work i ll try with openWARP. I ll keep you updated.

Thank you
Fabrizio

frrfrz, 2020-01-16:
Just one last question, i can't understand why using the same nemoh.cal input file (so same frequency range and step) and the float.dat and spar.dat from the tutorial  the result that i obtain are good. Thank you

TTToan2008, 2020-01-16:
If I understand correctly, you meant that you obtained a good agreement (no spike) between WAMIT and NEMOH using inputs from WEC-Sim repository. If so, which NEMOH version are you using? Did you include IRR function or other?

By the way, I just ran NEMOH simulation using exact **inputs from NEMOH folder** of WEC-Sim repo plus **your NEMOH mesh**. The result looks pretty good (attached figures) and it is similar to that of NEMOH result of WEC-Sim repo.
[image]

[image]

frrfrz, 2020-01-17:
Yes my problem is that if i run nemoh with the mesh from wecsim\tutorial\bemio\rm3 i obtain the results that you posted above, but if in the same folder i run nemoh with my mesh and then i run bemio i obtain different results.
[image]
[image]
(i've obtained these results running bemio from the folder that i sent you)
That's why i thought that the mesh was the problem. Same nemoh.cal, mesh.exe, preprocessor.exe, solver.exe, postprocessor.exe.
My nemoh version is the v2_03.

frrfrz, 2020-01-21:
P.S. I am running nemoh through the command window, i run mesh.exe separately fot the two bodies and then i just run prprocessor.exe, solver.exe and postprocessor.exe through a .bat file. You can find everything i use inside the folder i sent you.

TTToan2008, 2020-01-23:
Hi,

Sorry for my late response.
I think your mesh has an issue. As you look at meshes displayed in Meshmagick tool, your mesh resolution is really small at heave plate of spar part.  I think this is the reason why there are so many spikes in the results either NEHOH or WAMIT (WAMIT result showed in the previous file looks cleaner). Therefore, I would recommend that you should avoid different mesh resolution among discontinuous surfaces (e.g., heave plate side).  Additionally, you might want to consider a bigger mesh cell of spar closed to the water surface as well. 

Let me know if that helps.
Thanks,
Toan

[image]
[image]

frrfrz, 2020-01-27:
Hello Toan, with your suggestion everything works. I am not sure if the problem was caused just by the heave plate side, anyway i ve changed that mesh and also aligned the vertical lines of the spar mesh and float mesh.
[image]
[image]
[image]
[image]
Thank you so much for your time and help. You saved my master thesis.

Fabrizio

closed 2020-01-29 by frrfrz