WEC-Sim/WEC-Sim #334: only names in bodiesOutput

avidortelem, 2020-02-03:
Hello,

I am new to WEC SIM and I try to play a bit with the tool to understand how it works, I am reporting the following error that I do not understand because it is also links to the model that I would like to build.

I tried is to modified the OSWEC simulink model by:
- fixing the base to the flap (constraints 'Fixed', with the base as Base and the flap as Follower)
- doing a translational PTO of the FLAP relative to the seabed.

(the goal is then to make the system {BASE + FLAP} moving in surge relative to SEABED)

The only thing that I change in the wecSimInputFile is then the pto where I added:
pto(1).orientation.z=[1,0,0];   
for having the translation in SURGE.

When I run wecSim it looks to work (the system has well a surge movement in the visualisation)

until the error in responseClass.m row 40 , the bodiesOutput is only a 1*2 struct, there are only the names. (no time, no signals, etc...)

What could explain that the outputs are not here ?

Thank you very much.
Br Telem

kmruehl, 2020-02-19:
@avidortelem Thank you for your question and my apologies for the delay in response. Could you please upload a ZIP file with this WEC-Sim model so that I can resolve this issue? Thank you.

avidortelem, 2020-02-20:
Thanks !

I update an example in ZIP.

I use my nemoh outputs on a plate to run WECSIM with a surge constraints.

I use the flap geometry just for the visualisation.

I don't know if the error comes from my nemoh outputs or from something else..

Thanks again 
[model.zip](https://github.com/WEC-Sim/WEC-Sim/files/4230278/model.zip)

avidortelem, 2020-03-03:
Hi,

I did the simulation with hydrodata from wamit files and the error is still here...

Br T

avidortelem, 2020-03-03:
An other remark:

If I modified the OSWEC tutorial to make {FLAP+BASE} move surge according to global frame --> IT WORKS

If I built EXACTLY the same model but I do it manually from scratch --> it DOES NOT work 

Crazy..

avidortelem, 2020-03-10:
Hello @kmruehl do you have an idea when you look at the pictures above ?

Sorry to notify you but I really don't understand and I am block :/. I compared all the inputs (body, simu, waves, etc..)  when I run the simulations and there is no differences, so I really think the error comes from the simulink models. 

Br T

dforbush2, 2020-03-10:
Hi @avidortelem 

I'm new to your issue, but the little chain thing you have pointed out means that the block is "library linked" and it is referencing the WEC-Sim simulink library.

The fact that your "from scratch" simulink model does not have this could explain why your output file has only names if the link is somehow broken. I unfortunately cannot open your simulink model, as it is in 2019b. Could you please reattach with the model saved 2019a or earlier? See "Export a Model to a Previous Simulink Version". https://www.mathworks.com/help/simulink/ug/saving-a-model.html

As a short term solution, is it possible to copy/paste the needed blocks from tutorial and example files? They should be properly linked to the library and can be initialized and specified as needed from the WEC-Sim input files. I usually use this approach when I make a new model anyway.

avidortelem, 2020-03-11:
Hi @dforbush2 , 

Thanks for your answer.

I put enclosed a zip of the model.

There is several simulink models and we can see that according to the one we choose the simulation works or no:

- one simulink model made from OSWEC --> WORK
- the same one made from scratch --> DOES NOT WORK
- the same one made from scratch same in previous version (can you open it now ?) --> DOES NOT WORK
- the same one made by copying the block in a blank model --> DOES NOT WORK.

I think the 'link' thing was just an issue of display because finally I can see it now ...

dforbush2, 2020-03-11:
Hi @avidortelem ,

I was able to replicate your error using my own clone of the repository. Thank you for bringing this to our attention.

We have identified an issue with the library link that is causing this error, and are working on a fix. In the meantime, I would recommend copy/pasting the necessary blocks from a tutorial/example or applications case (https://github.com/WEC-Sim/WEC-Sim_Applications). Any block you might need should be found in one of those locations. Thank you for bringing this to our attention, and apologies for the delay in addressing your issue. 

I will reply once more and close this issue when the error has been resolved.

avidortelem, 2020-03-12:
Thank you very much ! I succeed to make it work now.

PS: 
It seems that it works when I use the simulink file of a tutorial, whatever if it is my block or not.
If I use a new blank simulink model and I copy the block from the tutorials, it doest not work.

Br T

nathanmtom, 2020-03-23:
@avidortelem Thank you for your patience awaiting our response to the issue you have raised. After further investigation, the WECSim team identified the issue is with the default parameters set in the "Model Configuration Parameters" by MATLAB. Starting in MATLAB 2019a a newly created .slx file will have a check box titled "Single Simulation Ouput" checked which causes the issue that you submitted with this issue request. If you un-check this option in the .slx file, save, and run WECSim the postProcess script should no longer have the issue you raised. 

The team has now added a check within the simulationClass.m to ensure that any new .slx files generated will have the option unchecked. If possible, please pull the latest version of WECSim from the Github website and check that a .slx file created from scratch will now execute properly. 

Please let us know if this hasn't solved your issue. 

Cheers, 
Nathan (National Renewable Energy Laboratory)

avidortelem, 2020-03-24:
Hello,

It seems to work now for me. Thanks you very much for your help :) !

closed 2020-03-24 by nathanmtom