WEC-Sim/WEC-Sim #336: MoorDyn on Linux

ttosdevin, 2020-02-06:
Hi,

I'm trying to run WEC-Sim with MoorDyn on a Linux machine and have followed the step suggested in issue #269 of changing line 88 in mooringClass.m to:
obj.moorDynInputRaw = textread('./Mooring/lines.txt', '%s', 'delimiter', '\n');

But I get the following error message:

 Running MoorDyn (v1.01.00C, 2016-04-20)
   Copyright (c) Matt Hall, licensed under GPL v3.
   Creating mooring system.  3 fairleads, 3 anchors, 3 connections.
   Error: unable to write file Line1.out
   Error: unable to write file Line2.out
   Error: unable to write file Line3.out
   Error: unable to write file Line4.out
   Warning from Catenary: Unstretched line length too large.
   Error: unable to write file Line5.out
   Warning from Catenary: Unstretched line length too large.
   Error: unable to write file Line6.out
   Warning from Catenary: Unstretched line length too large.
   Finalizing ICs using dynamic relaxation (5X normal drag)
    t = 1 s, tension at first fairlead is 77716.5 N    
    t = 2 s, tension at first fairlead is 1.01718e+06 N    
    t = 3 s, tension at first fairlead is 199041 N    
    t = 4 s, tension at first fairlead is 186692 N    
    t = 5 s, tension at first fairlead is 175286 N    
    t = 6 s, tension at first fairlead is 175517 N    
    t = 7 s, tension at first fairlead is 174200 N    
    t = 8 s, tension at first fairlead is 174005 N    
    t = 9 s, tension at first fairlead is 174790 N    
    t = 10 s, tension at first fairlead is 174934 N    
    t = 11 s, tension at first fairlead is 174871 N    
   Fairlead tensions converged to 0.1% after 11 seconds.        
Unable to write to output file 
Unable to write to output file 
Unable to write to output file 
Unable to write to output file 
Unable to write to output file 
Unable to write to output file   

I have tried this for 2 different WEC devices and on 2 separate Linux machines and I always get the above error (one of the cases is the RMS MoorDyn tutorial). The error appears to be from line 235 of the line.cpp MoorDyn file;
https://github.com/mattEhall/MoorDyn/blob/master/Line.cpp

Any help with this would be much appreciated.
Thanks,
Tom

yuyihsiang, 2020-02-10:
I noticed that you have "Mooring" instead if "mooring"

Not sure if that is the issue

Yi-Hsiang

ttosdevin, 2020-02-10:
Thanks for the quick reply, if I keep the folder name as "mooring" lower
case I get the below error, the same as from issue #269
"Error in 'RM3MoorDyn/MoorDyn': Initialization commands cannot be
evaluated."

The advice from that issue was to change the folder name to upper case and
replace any lower case "mooring" in reference to the lines.txt folder to
upper case "Mooring" in responseClass.m and mooringClass.m

yuyihsiang, 2020-03-04:
After spending some time with Matt, the moorDyn developer, we finally track down the issue and have a potential solution for you.

First, this is a moorDyn issue and only happens on Linux (not on a Mac). We were able to reproduce your issue. Unfortunately, we are unable to resolve this IO issue directly, but we will try to fix it later when we update the moorDyn for WEC-Sim.

Having said that, you should be able to run WEC-Sim/moorDyn on Linux by turning off the moorDyn output for each line. You still can get the load at the connection point from the Lines.out file.

Here is the snapshot of the updated lines.txt file. You basically change the chain Flags/Outputs" setting from "tp" to "-"

[image]
 
Hope this help,

Yi-Hsiang

ttosdevin, 2020-03-06:
Great, thanks!
Changing the Flags/Outputs resolved the issue.

Tom

closed 2020-03-10 by kmruehl