WEC-Sim/WEC-Sim #337: mass matrix

siya-jin, 2020-02-11:
Hi,
I am looking at an unsymmetrical hinged-raft device which has nondiagnonal elements in the moment inertia. 
It seems that in WEC-Sim the moment inertial can only be given for the diagonal term of Ixx Iyy and Izz. May i ask does WEC-Sim ignore the nondiagnonal elements? How about the nondiagnonal effects of added mass and radiation damping of Ry to X, Y and Z directions...

Thanks a lot

dforbush2, 2020-02-11:
Hello,
Because WEC-Sim solves body dynamics about the body center of mass, it is convenient to describe mass/inertia as a diagonal matrix in terms of principle inertias about principle axes X, Y and Z that are mutually perpendicular and coincident with the body center of mass.

However, the added mass and radiation damping matrices are 6 x (6*N), where N is the number of bodies included in the boundary element model. These matrices can (and usually do) include off-diagonal terms. 

Hope this helps.

siya-jin, 2020-02-11:
Hello,

Thanks for your information. Do you mean WEC-Sim do keep the non-diagonal terms of added mass and radiation damping? But referring to the bodyClass.m file, in the 'adjustMassMatrix' function, it seems that only the diagonal items of the added mass are merged into the mass matrix. This makes me confused..
From your experience, will ignoring the non-diagonal items of the inertia for a unsymmetrical body cause a big effect on the response of a unsymmetrical body.

Best wishes
Siya

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

dforbush2, 2020-02-12:
Hello,

Please see previous issue #161 as it may answer your question in more detail.

closed 2020-02-18 by kmruehl