WEC-Sim/WEC-Sim #339: Error Reading AQWA's .LIS and .AH1 Files

sugarl14, 2020-02-14:
Hi,

I have been performing an analysis on a rectangular box (just to get the hang of WEC-Sim), and I am having an issue with MATLAB reading the .AH1 and .LIS output files. I have read through the other similar issues that others have had with this problem, (such as #295  ) but still can't seem to figure mine out. I increased the number of wave frequencies to 50, with 4 total wave directions. Still, however, I am getting the error message below.  I've also attached my .ah1 and .lis files for you viewing if it helps. thank you!

[aqwa analysis.zip](https://github.com/WEC-Sim/WEC-Sim/files/4205761/aqwa.analysis.zip)

[image]

kmruehl, 2020-02-17:
@sugarl14 Thanks for attaching your AQWA files. We will look into this issue. It looks like you're running AQWA 18.1.00, is that correct?

sugarl14, 2020-02-17:
Version 18.1 is correct!

dforbush2, 2020-02-18:
Hi @sugarl14 ,

I had a look at your aqwa files. This version of AQWA has keywords like "COG" and "DRAFT" in commented sections of code. The Read_AQWA parser presently cannot distinguish between the commented remarks and the definitions of these quantities, and this leads to your error. While our approach has been robust for other versions of AQWA, it appears that it is not for 18.1.00. Thank you for bringing this to our attention. We will work on a patch to ensure compatibility with your version of AQWA.

sugarl14, 2020-02-18:
Thank you so much for your help! I look forward to hearing back from you soon.

-Landon

sugarl14, 2020-02-19:
Out of curiosity, what do you think the timeline for this bug fix would be? Also, what previous versions of AQWA are compatible with WEC-Sim? I may be able to obtain an older version from my university. 
Thanks!

Landon

dforbush2, 2020-02-19:
Hi Landon,

In this particular case, you would likely be better off either re-running AQWA such that your output files more closely resemble those found in the /examples/BEMIO/AQWA/Example folder and then running Read_AQWA.m. Unfortunately, I do not have access to AQWA so I can't offer much specific advice, but there are a large number of user-specified options in AQWA that affect output file structure even within a single version: perhaps tweaking these will give more easily parsed results?  

Your AH1 file is particularly unique and seems to be more like a read-me file on the AH1 format than an output of the AQWA run. From the perspective of Read_AQWA, it is incomplete, as it has only abbreviated outputs of added mass, radiation damping, and others. 

To process these particular output files, I would recommend writing your own version of Read_AQWA. Your .LIS file does appear to contain all necessary hydrodynamic information.

To do this, generally, you would emulate the syntax of lines 73 - 150 to populate the fields of hydro(F). (Issue #295 has some examples). But in your case, you should parse the .LIS file as that is the only one with fully populated outputs.

sugarl14, 2020-02-20:
Thank you so much @dforbush2 for your help!! I'll look more into that later on today and let you know if I come across any more issues. Thank you!

closed 2020-02-27 by dforbush2

sugarl14, 2020-03-04:
Hi @dforbush2,

I was able to acquire AQWA Version 16.2 from my university. Do you know if WEC-Sim be compatible with this version?

dforbush2, 2020-03-04:
Hi @sugarl14 

As mentioned, your AH1 file was really strange. I'm not sure that it was a version issue...more of a quirk of your particular run settings in AQWA. It's worth a try in another version. I recommend looking at the AH1 files in the BEMIO examples and seeing their general format, and trying to tailor the settings of your AQWA run to generate something similar.