WEC-Sim/WEC-Sim #340: Error in Test of Installation

NLWaves, 2020-02-17:
Hi There, I have the following error when running the installation test. I am attempting to run
>> cd $WECSIM/examples/RM3
>> wecSim
and then I get:

Failed to load library 'WECSim_Lib_2015b' referenced by
    'RM3/Float/Hydrodynamic Body/Kinematic Transforms/No
    B2B/nonLinYaw kinematic transforms'
    Error using wecSim (line 291)

Versions are as follows
WEC-Sim v4.0 (Latest version fromgit clone)
MATLAB Version: 9.7.0.1296695 (R2019b) Update 4
Operating System: Microsoft Windows 10 Education Version 10.0 (Build 18363)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
MATLAB                                              Version 9.7
Simulink                                              Version 10.0
Simscape                                             Version 4.7
Simscape Multibody                            Version 7.0

When I run slLibraryBrowser I can see the WEC-SIM objects lcoated in the window.
But I get the following warning in the main Matlab command window.
It seems to be a similiar error.

Warning: Unable to load block diagram 'WECSim_Lib_2015b' 
> In LibraryBrowser.internal.generateRepository
  In LibraryBrowser.internal.generateRepository
  In LibraryBrowser.internal.generateRepository
  In LibraryBrowser.internal.findLibraryInfo
  In LibraryBrowser.internal.findLibraryInfo
  In LibraryBrowser.internal.findLibraryInfo 

Any advice greatly appreciated.

dforbush2, 2020-02-18:
Hi NLWaves,

Thank you for your timely post!
The commit 3feefae on Feb 14th had a broken library link causing the error you described. 

This has been fixed by commit a187e57. Please pull the new update and let us know if you encounter further issues.

closed 2020-02-18 by kmruehl

NLWaves, 2020-02-20:
Thank you for your hard work
Its functioning perfectly now!