WEC-Sim/WEC-Sim #341: Applying control system to WEC-sim

bowdena3, 2020-03-04:
Hi,

I am new to WEC-sim and I wanted to know if it is possible to apply a control system to WEC-sim in the Simulink model, or does the BEM model need to be re-ran if i wished to change the damping for different wave inputs?

If I can apply control would I just add a controller subsystem linked to the PTO and is it possible to simulate latching control?

Thanks for the help!

dforbush2, 2020-03-04:
Hi @bowdena3 ,

Because WEC dynamics are solved in the time domain via Cummins formulation, the PTO force is calculated separately from the hydrodynamic coefficients determined from BEM, so I don't see a reason to re-run the BEM model for different values of PTO damping, provided that you have hydrodynamic coefficients calculated over a sufficient range and resolution for the simulated wave states.

We have several WEC-Sim Applications repository cases related to PTO modeling and control.
https://github.com/WEC-Sim/WEC-Sim_Applications

They should be a good starting point for modeling controllers of varying types and complexity. In particular, I would recommend the ./NonlinearControls and the ./PTO-Sim sub-directories. 

Best,
Dom

bowdena3, 2020-03-05:
Thank you for the help, this is really useful!

closed 2020-03-06 by dforbush2