WEC-Sim/WEC-Sim #342: Power extraction

frrfrz, 2020-03-08:
Hello,

I did some power extraction analysis using the Eureka scatter diagram from Reference Model Project and a BS spectrum, i ran a first set of test using the mooring lines from the MoorDyn tutorial, and a second set of test without mooring lines and with 6 DOF. I've obtained an AEP value of 708 MWh for the first set of test and 671 MWh for the second one. What this difference in power can be attributed to in your opinion?

yuyihsiang, 2020-03-10:
Hi,

It can be several reasons:
- Did you use waves.phaseSeed = <specific number>. If not MATLAB will randomly choose the random seed and will result in different irregular waves, which will lead to a different solution. The difference will depend on the length of the simulation time.
- Effect of mooring may also have some influence on the response of the system

Yi-Hsiang

frrfrz, 2020-03-14:
Hello Yi-Hsiang, 

thanks, as always, for your precious help, my waves charachteristics are the followings:
         
         waves = waveClass('irregular');         % Initialize Wave Class and Specify Type
         waves.H = 5.25;                         % Significant Wave Height [m]
         wavesT = waves_T_values(iteration_number);
         waves.T = wavesT;                       % Peak Period [s]
         waves.spectrumType = 'BS';              % Specify Wave Spectrum Type
         simu.ssCalc = 1;                        % Turn on State Space
         waves.freqDisc = 'Traditional';         % Uses 1000 frequencies

I've increased the simulation time to 1000s and to obtain a more reliable power result i've used a "for" loop to repeat the calculation 3 times and calculate the average value.
What do you think it's better to obtain a reliable power result and why?
I expected to obtain an higher power value running the simulation without mooring and with just 1 DOF but it didn't happen, what do you think about this?
Thank you so much.

Fabrizio

yuyihsiang, 2020-03-16:
I think 1000 s with 3 realizations should be fine. For consistency purposes, I would also recommend you set waves.phaseSeed = values(iteration_number); and this iteration_number changes for different sea states as well. 

For your question regarding AEP value, it is possible that 1DOF gives you lower AEP. Theoretically, 3DOF may let you obtain higher AEP. However, this depends on the cases.

Yi-Hsiang

frrfrz, 2020-03-16:
Ok, Thank you so much for your time!

Fabrizio

frrfrz, 2020-03-17:
Just one last doubt, you said: " it is possible that 1DOF gives you lower AEP. Theoretically, 3DOF may let you obtain higher AEP. However, this depends on the cases". Why do you think this happens? i would have expected to obtain an higher value of AEP without mooring in 1DOF, i thought about this as an ideal case.

yuyihsiang, 2020-03-18:
First, congratulations on your graduation and good luck.

Theoretically, more than one DOF can give you move power "if you can capture that", see Falnes' Ocean Waves and Oscillating Systems book in Chapter 6.

closed 2020-04-01 by kmruehl