WEC-Sim/WEC-Sim #343: limit range of motion

avidortelem, 2020-03-12:
Hello,

My current model is 

1 body ( a plate )  + 1 translational pto (surge)  + Seabed (global frame)

Thanks to the translational pto, the plate is already constraints in surge motion, it is what I would like so nice.

But I would like to limit this range of motion, so the plate can only move between x1=-0.5m and x2=0.5m for example.

I can imagine that there is a simple way of doing that but I can not find how.
The translational constraints does not have this option. Maybe by putting in serie some fixed and translational constraint blocks for example ?

Br T

yuyihsiang, 2020-03-16:
I think you can use the translation PTO joint with "force input and velocity output" and develop an end-stop controller block that connects to it to limit the motion.

Yi-Hsiang

avidortelem, 2020-03-16:
Hello @yuyihsiang 

Thank you for the help.

I am completely new to simulink so I am not totally sure.

Does "Translation PTO joint" = Translational PTO Actuation Motion ?

and "develop an end-stop controller block " is a subsystem that should be connected to it and develop to analysis position / velocity and changed it if necessary ? 

I put enclosed an picture with the my current model (translational PTO) and your suggested blocks at the right, is that what you meant :) ?

[image]

avidortelem, 2020-03-17:
Hello @yuyihsiang ,

Thank you very much for your help.

Is the figure below shows well what you suggested ?

If yes, I don't really know how to buildt the end stop controller.

Indeed what I would like to do is :

if x<xmin, then x=xmin and velocity=0.
if x>xmax, then x=xmax and velocity=0.

Then it would implied to modify position and velocity, but here the end stop controller output is the force, so no possibility of controlling directly the position.

[image]

Best Regards,

dforbush2, 2020-03-17:
Hi @avidortelem,

True endstops can be a little tricky to model because they are a sharp discontinuity in velocity, but we typically prescribe a force. In terms of algorithms, I would recommend having a large damper and spring with an engagement length just a little longer than where you want the hard endstop to be (using your example numbers: maybe x=-0.49 m on one end and x = 0.49 on the other). This will drastically slow and then rebound your device away from the endstop.

some pseudocode for one side that could live in a user-defined function block in your "End-stop controller block" would be like:

if x_position >0.49
F_spring= - K_stop (x_position - 0.49);
if x_velocity > 0
F_damp= - C_stop(x_velocity);
end
F_endstop=F_spring+F_damp;
end

Just make sure to format the output force appropriately for the PTO block and your degree of freedom.

Physically you can think of this as a stiff rubber piece that covers endstop. Numerically, this will give you a very close approximation to a hard stop, but will give you some flexibility in how the device will respond to hitting the stop. For example, will it bounce off elastically (if so set C=0) or will there be some absorbed energy as the device hits the stop? You can play with K and C to give you the desired dynamics. I don't know what your device mass or excitation is, but K should be a large number relative to the maximum inertia you expect in your model.

avidortelem, 2020-03-18:
Thanks very much  @dforbush2 ! 

Ok I did get that it was an additional force in the PTO block. Got it now and indeed it could be used for the end stop :).

My PTO is a linear one modeled with a stiffness Kpto =0 and a damping Cpto which depend on the wave but the value is usually around 10^5 Ns/m. The added mass of the plate is also around 10^5.

The plate hits the end stop (then some energy loss yes)  and does not bounce elastically, so I should put a really high C compare to Cpto ? and also a high K relative to the maximum inertia expected in the model  ? 
I will try to play with this numbers :).

closed 2020-03-18 by kmruehl