WEC-Sim/WEC-Sim #345: Calling wecSim in a function

ghleila, 2020-03-16:
Hello,

I'm trying to call wecSim in a function but it ends up with the following error:

 _**Error using wecSim (line 259)
    Variant control 'sv_b1_hydroBody' of variant block 'powerflow_v2/Rigid Body' must be a condition expression or name of a Simulink.Variant object in the global
    scope workspace of the model. Ensure that the condition does not match the model name.**_ 

This happens when wecSim is called in a function like case 2 below but it works in case 1 without any errors. 

case 1:
    initiallization;      
    wecSimInputFile;
     wecSim

case 2 :
function output=fcn1(input)

    initiallization;       
    wecSimInputFile;  
     wecSim

end 

Could you please help me with that?

Regards,
Leila

kmruehl, 2020-03-18:
@ghleila that error is related to the initialization of the variant subsystems in WEC-Sim. This looks  similar to issue #298 that you previously posted, that was resolved in issue #301. I think the problem is related to line 39 of `wecsim.m`. I recommend iterating through the diary log, similar to issue #301, or modifying line 39 of `wecsim.m`.

kmruehl, 2020-04-01:
@ghleila did this resolve your issue?

ghleila, 2020-04-01:
@kmruehl I tried modifying line 39  of wecSim.m but didn't work. Then I downloaded and installed the latest version of wecSim package and the problem solved. 
Thanks for your help!

kmruehl, 2020-04-01:
@ghleila wonderful! Glad the latest version of WEC-Sim resolved your issue.

closed 2020-04-01 by kmruehl