WEC-Sim/WEC-Sim #347: [RM3][desalination] Questions about instabilities in a customized model

lethai3, 2020-03-21:
Hi Wec-Sim team !

I hope everything is going well, despite the current Coronavirus situation...

I am trying to learn how to use WEC-Sim. More specifically, I wanted to model a reverse osmosis system with an absorber type RM3. (Most of the code has been naturally inspired by your recent desalination upload as well as the related NREL published papers).

Unfortunately, my solution presents an instability (see picture below) which I do not know how to solve. I guess the shape of the envelop makes some sense but I do not understand why the internal frequency is so high. Is it something that you have already  experienced in your simulation ? Do you eventually have an idea on how solve it (which parameters should I play with)? Feel free to provide me with any kind of insight !

[image]

I have attached the corresponding file if needed (see below)

Thank you very much for your work on WECSim as well as your potential consideration,

Stay safe and have a nice end of the week,

Remi 

[MyRM3_RO.zip](https://github.com/WEC-Sim/WEC-Sim/files/4362472/MyRM3_RO.zip)

dforbush2, 2020-03-24:
Hi @lethai3 

Is it possible to re-upload your model, exported as 2019a or earlier? I would like to help but I am unable to open your model.

https://www.mathworks.com/help/simulink/ug/saving-a-model.html#f2-86382

Best,
Dominic

lethai3, 2020-03-24:
Hi Dominic, 

Thank you very for your consideration!! Here is the model with the 2019 simulink version (called MyDesignRO_v2019).  Please tell me if you encounter other versioning issue ... 

Have a nice day and stay safe !

Remi

[MyRM3_RO_v2019.zip](https://github.com/WEC-Sim/WEC-Sim/files/4376015/MyRM3_RO_v2019.zip)

dforbush2, 2020-03-24:
It seems both of the uploads are missing a file, motorEff, that it needs to run?

lethai3, 2020-03-24:
Hi Dominic, 

Sorry to read this. I am a little confused because the program should not use “motorEff” to run, normally… Maybe it is because I only use the files located in this folder.

In the link below, I have taped the opening of the files on my computer, if needed. The quality is bad sorry by advance …
https://drive.google.com/a/berkeley.edu/file/d/14YIadfXBiXH8wjx8gvCsJmMP8VBFIuVE/view?usp=sharing

If you still have difficulties to open it, let me know but I should do because I am still confused about what what is going wrong…

Thanks for your time !

dforbush2, 2020-03-24:
I was able to make your model run by commenting lines 39 on of ptoSimInputFile.m and replicated your results. You must have had the necessary files (motorEff) to run the un-commented version somewhere in your matlab search patch, but you're correct in that these files are not needed to run the simulation.

Firstly, I would recommend that you re-visit the PTO-sim documentation...it seems you are trying to use the ptoSimInputFile to affect the elements you have modeled using the Simscape/Hydraulic blocks? This will not work. You will need to use the blocks in the PTO-Sim library to define your model in this way. I recommend the RM3_Hydraulic_PTO example in the WEC-Sim Applications repository as a starting point example https://github.com/WEC-Sim/WEC-Sim_Applications.git for this approach. If you've been trying a bunch of values by tweaking the ptoSimInputFile and getting the exact same outputs, this is the reason :) 

Alternatively, your model runs as is...but to change parameters of the components of your RO system you will need to do them manually from within the simulink model or programmatically, perhaps from your WEC-Sim input file. For this approach I recommend starting from the Desalination example in the WEC-Sim Applications Repository.

Nonetheless, your results are not an instability in the strict sense. The frequency you are seeing is essentially the time it takes for the one-way valve to crack open (occurs at 30 bar), and the combination of parameters (accumulator volume/pressures) with your selected wave state. At a high level, these hydraulic components are actuating and saturating several times per wave cycle, resulting in the higher frequency components you're seeing. I was able to smooth out your flow rate, for instance, by greatly increasing your accumulator volume and increasing the cracking and working pressure of the nearby one-way valve (manually, by changing their parameters in the simulink model itself, since changing them in ptoSimInputFile does not have an effect). I am not an expert in desalination, but it seems to me that the modeled wec is "oversized" for the modeled hydraulic system, since the latter saturates multiple times within a wave cycle. 

- Dominic

lethai3, 2020-03-24:
Hi Dominic,

**Thanks so much for your insight and your time!**

Yes, I should have highlighted earlier: I am not calling the ptoSimInputFile but rather a file called “value” (line 3 of the wecSimInputFile) to programmatically modify most of the parameters in the Simulink model. Sorry for the confusion. For sure, I will revisit the PTO Sim documentation.

Your description of the instable phenomenon makes sense to me. Then I guess it is a design problem rather a simulation one… 

I am just surprised about the fact that the modeled wec is “oversizing” the modeled hydraulic system since the modeled buoy volume is only 2 cubic meters. _In case you still have these values in mind, to what extent have you “greatly increased” the accumulator volume and the “cracking and working pressure”?_ 

Thank again for your help. I may let the issue remain opened a few days, **in case someone has more some design recommendation or any kind of insight.**

Have a nice day !

kmruehl, 2020-04-01:
@lethai3 it sounds like this issue WEC-Sim has been resolved. We cannot provide design advice, so I'm going to close this issue.

closed 2020-04-01 by kmruehl