WEC-Sim/WEC-Sim #348: Model visualization issue

zwjzju, 2020-04-08:
I'm a beginner at wec-sim.I made a simple model, which is composed of a cylindrical cylinder and a float. I set the cylinder to be fixed relative to the global coordinate system, while the float is translational relative to the cylinder.When I tried to simulate this model, I found that the model was out of place. I made several adjustments on Solidworks software, and the same problem occurred.I would like to know how to solve this problem. In addition, I would like to ask if there is any advice on the simulation setting of this model, because I often receive error messages during simulation. By adjusting the PTO damping value, the float either keeps still or flies away .I am looking forward to your reply.
![Uploading 
微信图片_20200408214933.png…]()

zwjzju, 2020-04-09:
[image]

kmruehl, 2020-04-15:
@zwjzju The body placement issue is likely due to the fact that the geometry must be output to `*.STL` with the origin at the body's cg. Would you be willing to upload a ZIP of your case directory for us to resolve this issue?

zwjzju, 2020-04-16:
@kmruehl Thank you very much for your reply.I think I've solved this problem with the correct *.stl file output settings.Now I have a new issue.I want the outer cylinder to be anchored to the water and the waterplane to pass through its geometrically symmetric plane (in the middle of the cylinder), but that didn't seem to be the case.In my results, the waterplane seems to automatically adjust to this position based on buoyancy.How should I set it up correctly?My original intention was to make the outer cylinder fixed, and the inner cylinder only moved in the direction of heave, but the simulation results did not meet the expectation, I think that the parameter setting or the simulink model is not correct.I will compress the file into a *.zip file and send it to you.Thank you very much!
[image]
[FixedOWC.zip](https://github.com/WEC-Sim/WEC-Sim/files/4484752/FixedOWC.zip)

kmruehl, 2020-05-13:
@zwjzju Can you clarify what you mean by: 
> I want the outer cylinder to be anchored to the water and the waterplane to pass through its geometrically symmetric plane (in the middle of the cylinder)

Do you want the cylinder anchored to the water? Or do you want the cylinder fixed in space, meaning it would be anchored at a fixed height relative to the seafloor?

closed 2020-05-27 by kmruehl