WEC-Sim/WEC-Sim #349: A Running problem of Wavestar.slx

Doudou-Li, 2020-04-08:
Hello,
A problem when I run the Wavestar.slx file in the Wavestar WEC Control Competition source code.
The problem is following:

Variant control 'sv_b_hydroBody' of variant block 'WaveStar/Arm(Frame, Float, BC, Motor)' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name.

I don't know what it is. Could you give me some advice on how to solve it. 

Many thanks.

nathanmtom, 2020-04-08:
@Doudou-Li  

Thank you for posting this issue about the WECCCOMP model. I have just pulled the WECCCOMP files from the following repository ``https://github.com/WEC-Sim/WECCCOMP" and was successful at running the model. 

Therefore, have you made any edits to the wecSimInputFile.m or the Wavestar.slx files? If possible, please send the folder with all of your files to see if we can reproduce your error.

Also, to clarify you have also downloaded the base WECSim package and have been successful at running the example files: https://github.com/WEC-Sim/WEC-Sim/tree/master/examples

Doudou-Li, 2020-04-09:
Dear all,

I think I find the problem. I didn't run the wemSim file before, cos it's not located in the Wavestar soure files. It's in WEC-Sim-master/source.

Thank you so much

BR

Doudou

closed 2020-04-09 by nathanmtom