WEC-Sim/WEC-Sim #350: error loading AH1 and LIS files from  AQWA output files

ignacio-byte, 2020-04-17:
i can´t convet the AH1 and LIS files to h5.files

clc; clear all; close all;
hydro = struct();

hydro = Read_AQWA(hydro,'aqwa_example_data.AH1','aqwa_example_data.LIS');
hydro = Radiation_IRF(hydro,15,[],[],[],[]);
%hydro = Radiation_IRF_SS(hydro,[],[]);
hydro = Excitation_IRF(hydro,15,[],[],[],[]);
Write_H5(hydro)
Plot_BEMIO(hydro)

i did not change the name in scrip instead i rename the "ANALYSIS" file to "aqwa_example_data"
i use matlab R2019a
ANSYS aqwa 2019R3
wecsim 4.0
i dont know if there is a problem whith the version of programs
but the thing is when i run the scrip i get this error

Unable to perform assignment because the indices on the left side are not compatible with the size of the right
side.

Error in Read_AQWA (line 201)
                hydro(F).sc_ma((kdiff-1)*6+1:kdiff*6,j,i) = tmp1;  % mag of scattering force

Error in bemio (line 4)
hydro = Read_AQWA(hydro,'aqwa_example_data.AH1','aqwa_example_data.LIS');

function hydro = Read_AQWA(hydro,ah1_filename,lis_filename)

% Reads data from AQWA output files.
%
% hydro = Read_AQWA(hydro, ah1_filename, lis_filename)
%     hydro ?         data structure
%     ah1_filename ?  .AH1 AQWA output file
%     lis_filename ?  .LIS AQWA output file
%
% See ??\WEC-Sim\tutorials\BEMIO\AQWA\...? for examples of usage.

[a,b] = size(hydro);  % Check on what is already there
if b==1
    if isfield(hydro(b),'Nb')==0  F = 1;
    else  F = 2;
    end
elseif b>1  F = b+1;
end

p = waitbar(0,'Reading AQWA output file...'); %Progress bar
e = 0;

hydro(F).code   = 'AQWA';
V182            = 0; % Set AqwaVersion flag "Version 18.2 or larger" to 0
tmp             = strsplit(ah1_filename,{' ','\','.'});
hydro(F).file   = tmp{length(tmp)-1};  % Base filename

fileID          = fopen(ah1_filename);
raw1            = textscan(fileID,'%[^\n\r]'); %Read/copy raw output, ah1
raw1            = raw1{:};
fclose(fileID);
fileID          = fopen(lis_filename);
raw2            = textscan(fileID,'%[^\n\r]'); %Read/copy raw output, lis
raw2            = raw2{:};
fclose(fileID);
N               = length([raw1(:);raw2(:)]);

% n=2;
n = 1;
TF = 1;
while TF == 1
    if isempty(strfind(raw1{n},'LHFR'))==0; V182 = 1; end % Change Aqwa Version flag to 1 if specific string is found in header
[... 148 lines cut ...]
            for i=1:hydro(F).Nf
                tmp1 = str2num(raw2{ln+6+(j-1)*(hydro(F).Nf+9)+(i-1)});
                if i==1;
                    ind = tmp1(1:3); tmp1(1:3)=[];
                else
                    ind = tmp1(1:2); tmp1(1:2)=[];
                end
                tmp2 = tmp1(2:2:end);
                tmp1(2:2:end)=[];
                hydro(F).sc_ma((kdiff-1)*6+1:kdiff*6,j,i) = tmp1;  % mag of scattering force
                hydro(F).sc_ph((kdiff-1)*6+1:kdiff*6,j,i) = tmp2;  % phase of scattering force
            end
        end        
        hydro(F).sc_re = hydro(F).sc_ma.*cos(hydro(F).sc_ph*pi/180); % real part of scattering force
        hydro(F).sc_im = hydro(F).sc_ma.*sin(hydro(F).sc_ph*pi/180); % imaginary part of scattering force
        lnlog(ln:ln+6+(j-1)*(hydro(F).Nf+9)+(i-1))=[ln:ln+6+(j-1)*(hydro(F).Nf+9)+(i-1)];        
    end
    
    d = floor(10*(ln+length(raw2))/N);  %Update waitbar every 10%, or slows computation time
    if (d>e) waitbar((ln+length(raw1))/N); e = d; end
    
end
hydro = Normalize(hydro);  % Normalize the data according the WAMIT convention
close(p);
end

kmruehl, 2020-05-13:
@ignacio-byte Can you please upload the `*.AH1` and `*.LIS` AQWA files, and the `bemio.m` file you're using? That will allow us to reproduce and debug this error. Thank you.

kmruehl, 2020-05-27:
@ignacio-byte can you upload the AQWA files for this case?

closed 2020-06-03 by kmruehl