WEC-Sim/WEC-Sim #351: Reference to non-existent field 'time'.

uzayr123, 2020-04-21:
Hi,
Attempting to simulate a point absorber (conical shape), the simulation works fine, but when hitting the post-processing and saving section, matlab throws the errors 👍 
"Reference to non-existent field 'time'.

Error in responseClass (line 40)
                obj.bodies(ii).time = bodiesOutput(ii).time;

Error in postProcess (line 68)
output =
responseClass(bodiesOutput,ptosOutput,constraintsOutput,ptosimOutput,mooringOutput,waves.type,waves.waveAmpTime,hspressure,
wpressurenl, wpressurel);

Error in wecSim (line 269)
postProcess

Not sure why this is occurring #250 seemed to run into something similar.

In the process of re-running nemoh with a greater range of wave frequencies, but this a shot in the dark more than anything 
[Wec Sim.zip](https://github.com/WEC-Sim/WEC-Sim/files/4512192/Wec.Sim.zip)
Have attached the relevant files if this helps

Cheers

dforbush2, 2020-05-05:
@uzayr123 

I believe that the issue described has been noted and handled in #334.

It is an issue with the default settings changing between matlab versions. Thankfully the fix is easy :)

Please reference the previous issue and let me know if  this does not solve your issue.

uzayr123, 2020-05-05:
I'll have a look to see if this fixes it, don't currently have access to my computer. As an update, I ended up having to restrict the the model to a translational pto for a project I was working on, this fixed the issue, not sure why.

Thanks for the info though

kmruehl, 2020-05-13:
@uzayr123 did changing the Simulink settings, similar to #334, resolve this issue?

closed 2020-05-21 by kmruehl