WEC-Sim/WEC-Sim #352: WECSim vs. ProteusDS rotational motion issue

mleary29, 2020-04-22:
Hello,

I simplified my WEC such that I can compare the motion of a floating cylinder with 6 DoF in Proteus and WECSim. For this, I am finding similar heave motion, but am particularly curious as to why the motion in pitch is much smaller in WECSim compared to ProteusDS.

Attached is:
1.my WECSim case (including a script to save output as WECSim_FloatOutput.mat), 
Note: The script within the WECSim case named WECSimOutputSave should be run after the case to obtain the output structure as WECSim_FloatOutput.mat for ProteusDS/WECSim Comparisons. 
2. Figures comparing WECSim/Proteus
3..the ProteusDS results
4. a script to compare the two. 

Thank you for your time and help.

Best,
Matt

[WEC_UUV_PTO_JustFloat.zip](https://github.com/WEC-Sim/WEC-Sim/files/4519368/WEC_UUV_PTO_JustFloat.zip)
[FloatFigures.zip](https://github.com/WEC-Sim/WEC-Sim/files/4519376/FloatFigures.zip)
[ProteusFloatResults.zip](https://github.com/WEC-Sim/WEC-Sim/files/4519379/ProteusFloatResults.zip)
[ProteusWECSimFloat.zip](https://github.com/WEC-Sim/WEC-Sim/files/4519381/ProteusWECSimFloat.zip)

dav-og, 2020-04-24:
Hi Matt,

Thank you for submitting the issue & attached data - I was able to run your WEC-Sim model and plotting scripts and reproduce the same results that you shared :)

I noticed some discrepancies in the roll and yaw results you shared (0 displacement in WEC-Sim and large displacements in ProteusDS), and I wanted to directly compare the yaw orientation (not just the velocity as you did). I modified your script to get the following plot:

[image]

It seems like your ProteusDS results have a very large yaw displacement - can you verify this is the case from your end?

Cheers,
Dave

mleary29, 2020-04-24:
Hi Dave,

Yes i receive the same Yaw displacement results. There are also disparities in surge and sway, which I believe is because in Proteus the float is drifting in regular waves in x and y, while in WECSim the float doesn't appear move much in x and y.

The yaw motion is a noted issue within ProteusDS, where things such as geometric curvature have 
been prone to cause unwanted yaw motion, which is often dealt with through an additional constant drag. 

 I don't intend to have an additional constant drag in this portion of comparison, and I am using this to see how WECSim/Proteus compare in a more dynamically simple case.

Best,
Matt

dav-og, 2020-04-24:
Hi Matt,

> Yes i receive the same Yaw displacement results.

In that case, I would advise focusing on resolving the numerical errors (positional drift) observed in your ProteusDS results before attempting a comparison between the two codes :)

As you noted, additional damping can be introduced...another common approach is to simply eliminate some degrees of freedom from the model.

You may also want to verify that the hydrodynamic databases (HDBs) provided to both models are consistent.

> I don't intend to have an additional constant drag in this portion of comparison

In the WEC-Sim model that you sent to me, you _have_ applied some [viscous damping](http://wec-sim.github.io/WEC-Sim/advanced_features.html#viscous-damping) - without this the results do seem to drift a little (albeit to a much lesser extent than observed in your ProteusDS results).

Cheers,
Dave

mleary29, 2020-04-29:
Hello,

Thanks for the update, I looked into it and I recharacterized my drag in both proteus and WECSim, and the comparisons look much better now in my opinion. Still fascinated as to why there is no motion in roll or yaw in wecSim, but overall these are comparing fairly well.

 Also, I am still figuring out how to correctly characterize drag in degrees of freedom outside of heave, for it seems like drag would be present but don't know what cd would be.

 My drag I have updated such that in WECSim my lines code for drag are:

%still trying to correctly characterize drag outside of heave motion
body(1).viscDrag.cd=[1 1 1.2 0 0 0]; 
body(1).viscDrag.characteristicArea=[1.83^2*pi/4 1.83^2*pi/4 1.83*0.813 0 0 0]; 

[JustFloat.zip](https://github.com/WEC-Sim/WEC-Sim/files/4553891/JustFloat.zip)
[ProteusFloatResults.zip](https://github.com/WEC-Sim/WEC-Sim/files/4553944/ProteusFloatResults.zip)
[WECSim_FloatOutput.zip](https://github.com/WEC-Sim/WEC-Sim/files/4553947/WECSim_FloatOutput.zip)
[ProteusWECSimFloat.zip](https://github.com/WEC-Sim/WEC-Sim/files/4553930/ProteusWECSimFloat.zip)

Thank you for the help!

Best,
Matt

dforbush2, 2020-05-14:
Hi Matt, 

Drag terms like this are frequently used as tuning parameters in simulations, since characterizing them exactly can be difficult. I usually start with something predicted by simple geometry as an initial guess and iterate from there, recognizing that drag coefficients in oscillatory flow can have distinct qualities (G.G. Stokes, "On the effect of internal friction of fluids on the motion of pendulums" (1851!))

Some of these can be parameterized by KC number
https://www.researchgate.net/publication/320844223_Hydrodynamic_Coefficients_of_Heave_Plates_With_Application_to_Wave_Energy_Conversion. 

If you have experimental data available for comparison (try Gaeta et. al "Implementation and Validation of a Potential Model for a Moored Floating Cylinder under Waves." 2020) , there are a few approaches available using free-decay and/or forced oscillation tests. 

https://www.sciencedirect.com/science/article/pii/S0029801818308151 (for the pitch mode, from FOSWEC testing)
https://www.sciencedirect.com/science/article/pii/S2214166917300656

Hopefully some of that is helpful. If you become aware of a good general approach to tuning these drag coefficients, that would be a major breakthrough. It seems to me that you should set your drag coefficients high enough to eliminate any motions you believe to be non-physical on less-important degrees of freedom.

closed 2020-05-21 by kmruehl