WEC-Sim/WEC-Sim #356: How to Calculation More than One Sea-state(e.g.Hs=2.5m, Te=8s)

duxiao96, 2020-05-31:
Hello, i am trying to do a validation of WEC-sim (for my master thesis) using the RM3 model and the data obtain from the following papers:

dforbush2, 2020-06-03:
Hi @duxiao96

1. I would recommend you check out WEC-SimMCR for this task. It allows you to specify a vector of sea-states which WEC-Sim will execute in order.
 https://wec-sim.github.io/WEC-Sim/advanced_features.html#multiple-condition-runs-mcr

There is an Applications example as well as a webinar for this feature, available at the provided link.

2. I would need more information on your particular model to answer this with certainty. If you have follow-up questions, please also attach your model. 

But as an example, the example/RM3 model approximates the PTO as a damper in heave only. The MECHANICAL power absorbed by the PTO is found in: 

output.ptos.powerInternalMechanics(:,3)

 where the third column is taken since the PTO applies to the heave degree of freedom. This is equal to the product of velocity and PTO-applied force: 

output.ptos.velocity(:,3) .* output.ptos.forceInternalMechanics(:,3). 

The forceInternal and powerInternal mechanics fields are taken because the specified damping is calculated inside the PTO block. If you have another force specified as an actuation force (i.e., you are using a block like WEC-Sim/PTOs/Translational PTO Actuation Force), then you would also have to include forceActuation in the latter product to calculate MECHANICAL power.

Given the subject of your investigation, you likely will want to examine ELECTRICAL power. If you are using blocks in PTOSim, then electrical power will output to output.ptosim. If you are using a custom PTO implementation, then I would recommend directly specifying your electrical power using a "To Workspace" block in simulink. 

However, if you are using your own custom PTO block, you can specify an output

duxiao96, 2020-06-10:
Thanks you very much.
Based on PTOsim, I also have some question.
1、The ElecPower is too small, several hundred watts.(regular Hs=2.5m,Te=8s)
2、the Phase A current is also small
However, I don't know where the simulation model is wrong or other problems.

Hi @duxiao96

https://wec-sim.github.io/WEC-Sim/advanced_features.html#multiple-condition-runs-mcr

duxiao96, 2020-06-16:
Sorry, I do not know how to I don't know how to attach the file.
Can I send it to your mailbox?
At the same time, the reply mail last time has a model.

dforbush2, 2020-06-16:
Hi @duxiao96 

You should be able to attach a file by dragging/dropping into the comment box on git. But if you are talking about the unmodified PTO-Sim/RM3_DD_PTO example in the WEC-Sim Applications repository I can see this model already. 

Looking through the model, everything seems to be in order. An electrical output of several hundred watts does not seem unexpected. Parameters involved in that calculation can be varied within the `ptoSimInputFile.m` if you'd like to investigate other combinations. I would also try sea states with a different energy period if you would like to see higher values.

I would recommend using the Fourier transform on your fully-developed output data to get voltage or frequency content. Relevant output data is in output.ptosim.pmLinearGenerator. If needed (for example) for a control system it is also possible, albeit more challenging, to do this in "real-time" within the model. I highly recommend writing your own blocks to do this, but there are some pre-made options available in Simscape Electrical and DSP toolboxes.

kmruehl, 2020-07-08:
@duxiao96 does this resolve your issue?

duxiao96, 2020-07-13:
thank you very much!

closed 2020-07-13 by duxiao96