WEC-Sim/WEC-Sim #357: Body motions between bodies

jackhughes1, 2020-06-01:
Hi, hope you're doing well in these strange times. 

I have a question about body motions, in the example that the body is semi-submerged and is connected by a body above the waterline, such as a catamaran, would you be able to suggest a way to best model this in the Simulink model? 

So far I have included multiple bodies in the model but they all move as one under wave loading, rather than moving under independent wave loading and motions affecting each other. 

Sorry if that is difficult to comprehend, it's difficult to explain. But please don't hesitate to ask me for more information. 

All the best,
Jack

dforbush2, 2020-06-03:
Hi Jack.

I am envisioning exactly a catamaran, wherein two semi-submerged hulls are connected by a completely out-of-the water rigid platform, so if that is inaccurate, feel free to ignore the forthcoming. If you are able, it may be best if you could provide your model so that we are all on the same page. 

My first guess would be to model each hull as hydrodynamic bodies (such that they are subject to wave dynamics), and the connecting platform as a non-hydrobody (such that it is NOT subject to wave dynamics, and only moves as a result of the rigid coupling to the hulls). This will allow you to fix the spacing of the submerged hulls using another body that is "out of the water". 

Secondly, you will have to turn the body-to-body interactions flag ON (simu.b2b=1; in wecSimInputFile) if you wish the body-initiated waves of one hull to affect the other. This requires that you have BEM data quantifying this interaction.

kmruehl, 2020-06-03:
@jackhughes1, I want to echo everything @dforbush2 has already stated, and also point you in the direction of the WECCCOMP WEC-Sim model of the WaveStar device. It has many of the features you need: a fixed platform above the water surface modeled as a non-hydro body, and a hydro-body moving relative to the fixed platform. I've also attached a reference describing this model. 

https://github.com/WEC-Sim/WECCCOMP
[2018 - OMAE2018_Tom et al. - Numerical Model Development and Validation for the.pdf](https://github.com/WEC-Sim/WEC-Sim/files/4726567/2018.-.OMAE2018_Tom.et.al.-.Numerical.Model.Development.and.Validation.for.the.pdf)

jackhughes1, 2020-06-04:
Thanks for the prompt response! I'll have a look through this material and get back to you.

jackhughes1, 2020-06-11:
Hi @dforbush2, 

Yes, resolved. The example provided was very useful, thanks a lot. 

Jack

closed 2020-06-11 by dforbush2